Moddable-OpenSource/iot-product-dev-book #1: Connection Refused when flashing the mods with mcrun command

GobbinKorneel, 2020-08-17:
I'm using Ubuntu 16.04 (on Oracle VM Virtual box) and at first everything is working well. I can install the host and the corresponding modules on an ESP32. But after a while, it is no longer possible to do so. When I install an example with mcrun, I get the following error: 
[image]
When I try to reinstall the host, it seems to write to the ESP32, but in the console of xsbug, it traces content of older, earlier installed programs. 
I tried to erase completely the flash of the ESP32 but that doesn't solve it.
When I install other examples with mcconfig, like the ones shipped with the Moddable SDK, it works fine (helloworld). 
HELP!!!
The reason I use Ubuntu is that I don't succeed to program an ESP32 on Windows. To bad there are no video's to help newbies like me with that.
Thanks.

phoddie, 2020-08-17:
The "Connection refused" error is coming from [here](https://github.com/Moddable-OpenSource/moddable/blob/6729c9482b9186f3654d8b158f095451edb74e62/tools/serial2xsbug/serial2xsbug_lin.c#L115). That is an attempt to connect to xsbug (the debugger) by the serial2xsbug tool.

There are a few reasons that the connection attempt might have been refused:
1. xsbug has hung
1. there is more than one instance of xsbug running
1. a firewall is preventing the connection

Since your connections have worked, the firewall (3) seems unlikely. Check your system to see if xsbug is running (visible or in a hidden task). If you find any, terminate those and then retry.

> Note: The serial2xsbug tool bridges the serial communication from your ESP32 to the xsbug, which communicates over a network socket. When you build using `mcrun` it automatically launches serial2xsbug after installing the software, so that you can debug.

phoddie, 2020-09-02:
Please let us know if the suggestions above helped solve your problem. Thank you.

GobbinKorneel, 2020-09-10:
This solved the problem. Thank you! The reason was that more than one instance of xsbug was running.

closed 2020-09-10 by GobbinKorneel