Moddable-OpenSource/iot-product-dev-book #2: mcrun command problem?

LongNguyen1984, 2020-09-28:
Why when I use 'mcrun' command to build for esp32, it always returns to
###TypeError: cannot coerce undefined to object

phoddie, 2020-09-28:
Sorry to hear that. Could you let me know what directory / project you are building so I can check the manifest? Thank you.

LongNguyen1984, 2020-09-28:
I placed on these path:
%userprofile%ESP32/moddable; 
%userprofile%ESP32/esp-idf;
%userprofile%ESP32/iot-product-dev-book

phoddie, 2020-09-28:
I'm not sure I understand. Perhaps you are referring above to your system's PATH variable?

Here is are the commands for using `mcrun` to run the `helloworld` example
```
cd %EXAMPLES%\ch1-gettingstarted\helloworld
mcrun -d -m -p esp32
```

Did you follow these steps? If you did something else, what did you do?

LongNguyen1984, 2020-09-28:
Sure, I have followed this step with another command to build the host first:
> cd %EXAMPLES%\ch1-gettingstarted\host
> mcconfig -d -m -p esp32

This step worked well but  in the next step, I use 
> cd %EXAMPLES%\ch1-gettingstarted\helloworld
> mcrun -d -m -p esp32

it gave error:
###TypeError: cannot coerce undefined to object

phoddie, 2020-09-28:
Thank you for the clarification.

The error you report (TypeError) most commonly occcurs when the manifest file is not valid JSON. 

The file at `cd %EXAMPLES%\ch1-gettingstarted\helloworld` should look like this:

```json
{
	"modules": {
		"*": "./example"
	}
}
```
Is that what you see?

LongNguyen1984, 2020-09-28:
Yes, I have checked the file and it has the same content as you mentioned.

phoddie, 2020-09-29:
It looks like this is only a problem on Windows. We have a patch. You can try it. You need to replace two lines in the Moddable SDK -- [mcmanifest.js lines 312 and 313](https://github.com/Moddable-OpenSource/moddable/blob/e1fe8b4909e5a0518dc7b04e56caea1ab73ea23e/tools/mcmanifest.js#L312-L313) with the following:

```js
				if (sdkconfigFile  !== undefined){
					let sdkconfigFileMinGW = sdkconfigFile.replace(/\\/g, "/");
					tool.setenv("SDKCONFIG_FILE_MINGW", sdkconfigFileMinGW);
				}
```

After making the change, rebuild the Moddable SDK tools:

```
cd %MODDABLE%\build\makefiles\win
build
```

If you can try that, please let me know if it helps. Thank you!

We will push this change to the Moddable SDK in the coming days.

LongNguyen1984, 2020-09-29:
Dear Hoddie, 

I have tried by replacing the code you mention, and now it returns as info below:

> C:\Users\nmhoa\ESP32>moddable\tools\serial2xsbug\serial2xsbug_win.c(135): The system cannot find the file specified.
> NMAKE : fatal error U1077: 'C:\Users\nmhoa\ESP32\moddable\build\bin\win\debug\serial2xsbug.EXE' : return code '0x1'
Stop.

LongNguyen1984, 2020-09-29:
This is the structure of folders in ./bin and ./tmp after I tried to rerun 'mcrun' command:

In Folder ./bin
C:
├─config
├─helloworld
│      helloworld.xsa
│
└─host
        bootloader.bin
        partition-table.bin
        xs_esp32.a
        xs_esp32.bin
        xs_esp32.map
and in folder ./tmp/helloworld
C:.
│  check.js
│  makefile
│  mc.format.h
│  mc.rotation.h
│  mod.config.js
│
└─files
    │  check.xsb
    │  example.xsb
    │
    └─mod
            config.xsb

lprader, 2020-09-29:
The error that you're getting means that serial2xsbug can't find the COM port specified by the `UPLOAD_PORT` environment variable. This may be because:

1. The `UPLOAD_PORT` environment variable isn't set or is set incorrectly
2. The COM port doesn't exist because the device isn't plugged in
3. The COM port doesn't exist because the driver isn't installed

Can you double check that you completed all the steps in the ESP32 setup instructions in the Moddable SDK [Getting Started guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp32-windows)?

LongNguyen1984, 2020-09-30:
@Iprader: I can run the code with 'mcconfig -d -m -p esp32' without any problems from the loader that automatically detects COM port, but for the 'mcrun' command, it is impossible now.

phoddie, 2020-09-30:
Correct. `mcrun` does not  automatically detect the COM port. You must correctly configure it as @lprader described.

LongNguyen1984, 2020-09-30:
I have just added the UPLOAD_PORT environment variable and great to see it works now. However, it needs to open 'xsbug' first; otherwise, it will go into error as below:
No connection could be made because the target machine actively refused it.
> C:\Users\nmhoa\ESP32\moddable\tools\serial2xsbug\serial2xsbug_win.c(107): No connection could be made because the target machine actively refused it.
> NMAKE : fatal error U1077: 'C:\Users\nmhoa\ESP32\moddable\build\bin\win\debug\serial2xsbug.EXE' : return code '0x1'
Stop.

lprader, 2020-09-30:
Thanks for the report. We've been able to reproduce the problem on Windows and we'll take a look at a fix. For now, you can just run xsbug manually before running `mcrun`.

LongNguyen1984, 2020-10-01:
I have just found one more problem with 'mcrun' that it is impossible to load ssid and password from the command. So for the command like:
> mcrun -d -m -p esp32 ssid="my wifi" password="--------"
it can not generate config file and the solution that is I must build host with said and password first through 
> mcconfig -d -m -p esp32 ssid="my wifi" password="--------" for the host file first and using 'mcrun' in simple form later as following
> mcrun -d -m -p esp32

phoddie, 2020-10-01:
Hello, again!

> #> mcrun -d -m -p esp32 ssid="my wifi" password="--------"

Your command line is correct. However, it seems possible that your code is looking for those values in the wrong place.

The configuration arguments for the host, built with `mcconfig`, are the `config` module.

The configuration arguments for the mod, built with `mcrun` are in the `mod/config` module. Here is an example of accessing those:

https://github.com/Moddable-OpenSource/iot-product-dev-book/blob/aad4ae26c07f89c02a6180d7013cbcd61460df7b/ch3-network/host/main.js#L25-L27

LongNguyen1984, 2020-10-02:
I don't know why when I try to rebuild the host without ssid and password, it always read the check.js files rather than stop at the first condition in the main.js as here:

> if (!LoadMod.has("check") || !LoadMod.has("example"))
		return trace("Host installed. Ready for mods.\n");
Enventhough I have run  
> mcconfig -d -m -p esp32 -t clean 
to delete both old host and example files.

lprader, 2020-10-02:
The `-t clean` option does not delete host and example files from the device. It deletes build outputs for the app from the `$MODDABLE/build` directory. For more details, see the `mcconfig` section of the [manifest documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md).

If you need to erase the flash on your device, you can use [esptool](https://github.com/espressif/esptool). For example:

```
cd PATH\TO\esptool
esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 erase_flash
```

LongNguyen1984, 2020-10-03:
>  if (LoadMod.has("mod/config")) { 
>  	const config = LoadMod.load("mod/config"); 
>  	if (config.ssid) {

Dear Hoddie and Iprader,  
I also try to build the host as in the example of Chapter 3 but it can not detect the ssid and password.

phoddie, 2020-10-03:
@LongNguyen1984 - it is difficult to diagnose the problem without knowing precisely what you did and what happened. Would you kindly provide the command line (or command lines) you entered and indicate what happened after each? Thanks you.

LongNguyen1984, 2020-10-03:
I simply build the host in chapter 3 first with command:

> mcconfig -d -m -p esp32

and the debuger is activated and output with the text: 

"No Wi-Fi SSID"
C:\Users\nmhoa\ESP32\iot-product-dev-book\ch3-network\http-get-json\example.js (15) # Break: breakpoint!

Then, next I move to the example get-json and try command:

> mcrun -d -m -p esp32 ssid="my wifi" password="my pass"
But the debugger returns the same output:

"No Wi-Fi SSID"

>

phoddie, 2020-10-03:
The "No Wi-Fi SSID" message is a little confusing, but it doesn't indicate a problem here. It is specifically referring to the Wi-Fi configuration of the host -- and you built without one. The host uses the Wi-Fi configuration you passed to `mcrun`.

I committed a [change](https://github.com/Moddable-OpenSource/iot-product-dev-book/commit/81c99a656344727b5bebd76a56bdba530d657e19) to the Chapter 3 host to eliminate the warning.

After you update and repeat the steps above you should see something like this in xsbug:

```
Wi-Fi trying to connect to "my wifi"
Wi-Fi connected to "my wifi"
IP address 192.168.1.30
<!doctype html>
<html>
```

LongNguyen1984, 2020-10-03:
I have tried to modify the files as you mention but it output the same result:

> No Wi-Fi SSID

when I run the command: 

> mcconfig -d -m -p esp32

phoddie, 2020-10-04:
If your build is tracing "No Wi-Fi SSID" and it is not tracing "Wi-Fi bring to connect...", then it seems you are not running with the changes. You can try reinstalling a clean build of the host:

```
cd (whatever)/iot-product-dev-book/ch3-network
mcconfig -d -m -p esp32 -t clean
mcconfig -d -m -p esp32
```

LongNguyen1984, 2020-10-05:
Dear, Hoddie! 

I have tried to clean the old output first, then rebuild but it always runs into the debugger with "No Wi-Fi SSID" as I have mentioned above. 

> I don't know why when I try to rebuild the host without ssid and password, it always read the check.js files rather than stop at the first condition in the main.js as here:
>
> if [...]

LongNguyen1984, 2020-10-05:
I tried to delete the moddable SDK and rebuild it. Then, I rerun the command:

`

> mcconfig -d -m -p esp32 

`
The program will go into "No Wi-Fi SSID", and if I includes the ssid and password,
it will go into the example http-get-json I have builtt before. 
I think the compiler has stored the old files somewhere but I do not know.

phoddie, 2020-10-05:
Apologies. The change to eliminate "No Wi-Fi SSID" was only made to for ESP8266. This [commit](https://github.com/Moddable-OpenSource/iot-product-dev-book/commit/c29829645b8fcc97bf3483ba2a526c75a1ce3ef3) applies it to both ESP8266 and ESP32.

LongNguyen1984, 2020-10-06:
Thanks Hoddie, 

Now the command:
> mcconfig -d -m -p esp32

go into the first condition and return:

> "Host installed. Ready for mods."

Now, I move back to the example http-get-json try the command:

> mcrun -d -m -p esp32 ssid="my wifi" password="my pass"

and it runs and goes into the gdb as the output in the command line window below:

> ELF file SHA256: 0f71b08e42f7ca4a0eab2be3f554155d61ab950a94d8fd039c40c22471447042
> Backtrace: 0x400887f4:0x3ffbec80 0x40088a35:0x3ffbeca0 0x400d2983:0x3ffbecc0 0x40144717:0x3ffbecf0 0x4013a361:0x3ffbed20 0x4012a3fc:0x3ffbed90 0x40113ab7:0x3ffbef30 0x401083ec:0x3ffbefa0 0x40107f35:0x3ffbf040 0x40108096:0x3ffbf0d0 0x401339f9:0x3ffbf160 0x40113cd1:0x3ffbf200 0x400e2265:0x3ffbf270 0x400e068e:0x3ffbf290 0x400e06eb:0x3ffbf320 0x400e0f01:0x3ffbf340 0x400d2031:0x3ffbf4f0 0x400d203b:0x3ffbf540 0x40088d26:0x3ffbf560
> Entering gdb stub now. 
> 

So what should I do now next?

lprader, 2020-10-06:
There was a bug with `mcrun` on Windows. [This commit](https://github.com/Moddable-OpenSource/moddable/commit/2ac0cc0f4be4d23b2b9accd8c5dca0c9cc25a3ff) should fix it; can you update the Moddable SDK, try again, and let me know if it works now?

LongNguyen1984, 2020-10-07:
Dear Iprader,

The situation does not change even worse because now for both commands  'mcconfig' and 'mcrun' the output is always as below:

> ELF file SHA256: 434724d735bfb8817179433cb53bf598a78d925aef2f98939ae859029f74b7dc
> Backtrace: 0x400887f4:0x3ffbee80 0x40088a35:0x3ffbeea0 0x400d2983:0x3ffbeec0 0x40144717:0x3ffbeef0 0x4013a361:0x3ffbef20 0x4012a3fc:0x3ffbef90 0x40113ab7:0x3ffbf130 0x401083ec:0x3ffbf1a0 0x40107f35:0x3ffbf240 0x40108096:0x3ffbf2d0 0x401339f9:0x3ffbf360 0x40113cd1:0x3ffbf400 0x400e2265:0x3ffbf470 0x400e068e:0x3ffbf490 0x400e06eb:0x3ffbf520 0x400e0f01:0x3ffbf540 0x400d2031:0x3ffbf6f0 0x400d203b:0x3ffbf740 0x40088d26:0x3ffbf760
> Entering gdb stub now.
>

lprader, 2020-10-07:
Can you try erasing the flash on the device and doing a clean build of the host and example?

LongNguyen1984, 2020-10-07:
Dear Prader,

After erasing my device flash, the command 
> mcconfig -d -m -p esp32 

return to 'Host installed. Ready for mods.' first time. 
But then I run the code:
> mcrun -d -m -p esp32 ssid="my wifi" password="my pass"
 it returns the situation like before. 
Now I turn back to the host and rerun the code:
> mcconfig -d -m -p esp32
It will return to the same output: ".... Entering gdb stub now"

LongNguyen1984, 2020-10-07:
Again, I dont see the condition in the file main.js activate:

if (LoadMod.has("mod/config")) {

...}

lprader, 2020-10-08:
Odd. I'm trying to reproduce the bug, but haven't been successful since that update. Can you confirm that you took the following steps?

1. Update the Moddable SDK and do a clean build of tools. We have [this document](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Keeping%20Up%20To%20Date.md#host-windows) to guide you through that if you need it.
2. Erase the flash on the device using [esptool](https://github.com/espressif/esptool).
3. Do a clean build of the host.
4. Do a clean build of the example.

LongNguyen1984, 2020-10-09:
Hi Iprader,
I reupdated the Moddable SDK and rebuilt the host and example files and now the condition:
if (LoadMod.has("mod/config")) {

...}
activate now and return the output as below:

> Wi-Fi connected to "my wifi"
> IP address 192.168.0.25
> XS abort: not enough keys
> C:\Users\nmhoa\ESP32\moddable\xs\platforms\esp\xsPlatform.c (226) # Break: C: xsDebugger!

It stops and not load the example

phoddie, 2020-10-09:
@LongNguyen1984 - thank you for the update.

It seems likely that you did not update your copy of this repository. The following code provides ample keys:

https://github.com/Moddable-OpenSource/iot-product-dev-book/blob/c29829645b8fcc97bf3483ba2a526c75a1ce3ef3/ch3-network/host/manifest.json#L8-L12

That said, you have not indicated the example you are running here. Therefore, @lprader and I can only guess what might be going on.

LongNguyen1984, 2020-10-09:
Thank Hoddie, 

It works now. So should I reupdate all examples in /iot-product-dev-book/ ?

phoddie, 2020-10-09:
It is probably a good idea to update the entire repository so that you have the most recent versions.

phoddie, 2020-10-15:
Closing issue as it appears all the topics raised within it have been resolved.

@LongNguyen1984 -- many thanks for your patience in working through these!

To thank you for your help., we would be happy to send you a Moddable t-shirt. Just send us an email to info@moddable.com with your size and address.

closed 2020-10-15 by phoddie

LongNguyen1984, 2020-10-16:
Thank you Hoddie,

Great to receive the gift from your team.