Moddable-OpenSource/iot-product-dev-book #3: [Bug] Problems running mcconfig due to improper

ariznaf, 2020-11-10:
Build environment: windows 10
target: NodeMCU ESP32 (ESP-WROOM-32)

Description:
I have bought this book and followd instructions to install moddable and esp-idf in my computer.

I can run some moddable/examples as the helloworld using mcconfig -m-d -p esp32

When trying to flash the host code using examples in your book, I get errors.

For example in ch1-gettingstarted/host when I try mcconfig -d -m -p esp32 I get error in the manifest.json file

D:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-ble\host>mcconfig -d -m -p esp32
D:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-ble\host\manifest.json(17): error: missing name!
### Error: 'D:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-ble\host\manifest.json': invalid manifest!

The problem seems to be that the manifest file is not properly formatted.

There are trailing ',' in maps that have just one key/value pair that cause the problem

	"build": {
		"NAME": "ch1host",
	}

It works when I change it to:
	"build": {
		"NAME": "ch1host"
	}

In all the examples there are this kind of errors, with trailing ',' at the end of a map key/pair that cause parsing errors.

It works when you delete the trailing ',',

phoddie, 2020-11-10:
Thanks for purchasing the book! Apologies for the trouble.

A recent change to the Moddable SDK causes JSON with trailing commas to be rejected. Apparently we forgot to check the manifests in this repository after that change. We'll take care of that shortly.

Thank you for taking the time to report this issue. The fixes will benefit all of our readers.

ariznaf, 2020-11-10:
OK.
Vscode highlights the trailing , as an error, so it is easy to fix it.

lprader, 2020-11-10:
@ariznaf -- I just committed changes that remove the trailing commas in manifests. Thanks for reminding us we needed to do that ;)

closed 2020-11-10 by lprader

ariznaf, 2020-11-10:
Please, let me know how I can upgrade my files (using git).

lprader, 2020-11-10:
[This webpage](https://confluence.atlassian.com/bitbucketserver/basic-git-commands-776639767.html) has a good summary of basic git commands.

If you haven't made any local changes, use the following command to update: 

```
git pull
```

If you have local changes that you'd like to save, you should [stash](https://git-scm.com/docs/git-stash) your changes first, then replace them after you pull. For example:

```
git stash
git pull
git stash pop
```