Moddable-OpenSource/iot-product-dev-book #4: [Question] How do you run the examples?

ariznaf, 2020-11-10:
I am using moddable in a windows 10 machine, and a target device of type ESP32 ESP-WROOM-32 Devkit C (one puschased from AZ_Delivery).

I can run the basic moddable/examples without problems.

I am having problems running the ch1-gettingstarted/helloworld demo.

I can flash the host using 'mcconfig -d -m -p esp32' and I get 'Device flashed. Ready to install apps.' message in the console of xsdebug.

Then I close the debugger and try to run the demo app in helloworld using 'mcrun -d -m -p esp32' 

It compiles the app and links it, then it runs xsbug, but nothing happens

It does not go to the 'debbuger;' line in the code nor prints 'hello world' in the console.

What I am doing wrong?

phoddie, 2020-11-10:
Because it traces "Ready to install apps.", You have successfully installed the host for Chapter 1. That's great -- that's the most complex operation.

I just tried the same example here and it worked as expected.  I did that on macOS. What computer OS are you using?

FWIW - you don't need to quit xsbug. You can leave it running between sessions. In the terminal window, use Control C to disconnect the device from the debugger.

ariznaf, 2020-11-10:
Yes the host works flashing works OK.

I run mcconf -d -m -p esp32

and it opens the xsbug and get the message in the console log.

Then I keep xsbug open and use Ctrl+c to stop the mcconfig.bat script

The problem comes when trying to run the helloworld app in the corresponding dir.

I run
mcrun -d -m -p esp32 

It print again the Device flashed. Ready to install apps.' message, but nothing happens.

It does not even open the example.js file in the debugger, and I have no option in the debugger to pause the app or anything, aund just a blank screen.

When I run the moddable/helloworld example it opens the source code and stops in the first line.

I am using Windows 10 (i a standard PC).

phoddie, 2020-11-10:
Apologies, I overlooked your statement above that you are using Windows 10.

Perhaps, for some reason, the USB connection to your ESP32 is not COM3. Because of differences in `mcconfig` and `mcrun`, you might be able to flash a host but not install an example. The bottom of page 15 and top of page 16 have a trouble shooting note for Windows 10. Take a look? Perhaps you need to configure the `UPLOAD_PORT` as it shows.

ariznaf, 2020-11-10:
No, It is not COM3, in my case is COM7.

But I have stablished UPLOAD_PORT environment variable to COM7.

I have taken a look to the trouble shooting pages...
But I don't have any error of type 'warning: serialport_set_baudrate: baud rate 921600 may not work'.

After changing baud rate, should I recompile mcconfig? How? (it is not explained in the book).

When I use mcconfig to build the host, I messages of uploading, something like this:

```
d:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-gettingstarted\host>mcconfig -d -m -p esp32
        1 archivo(s) copiado(s).
Checking Python dependencies...
Python requirements from d:\users\fernando\ProyectosProg\esp32\esp-idf\requirements.txt are satisfied.
Running ninja in directory d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\idf
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[3/3] Generating xs_esp32.bin
esptool.py v2.8
Running esptool.py in directory d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\idf
Executing "C:\Python27\python.exe d:\users\fernando\ProyectosProg\esp32\esp-idf\components/esptool_py/esptool/esptool.py -p COM7 -b 921600 --after hard_reset write_flash @flash_project_args"...
esptool.py -p COM7 -b 921600 --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 xs_esp32.bin
esptool.py v2.8
Serial port COM7
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:62:ab:cb:00:e4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 18432 bytes to 11977...
Wrote 18432 bytes (11977 compressed) at 0x00001000 in 0.1 seconds (effective 1024.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 2457.6 kbit/s)...
Hash of data verified.
Compressed 675184 bytes to 377704...
Wrote 675184 bytes (377704 compressed) at 0x00010000 in 5.7 seconds (effective 949.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...

```

But when I use mcrun to upload the app it does not seem to try to upload anything:
```
d:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -p esp32
# xsc example.xsb
"# xsl "helloworld".xsa"
```

phoddie, 2020-11-10:
Because `mcconfig` and `mcrun` are separate tools, their output is quite different Your output from `mcconfig` is helpful -- it confirms that you are using COM7, not COM3.

The baud rate warning can be ignored. While it is sometimes an issue, the fact that you are able to flash to the host indicates there is no problem with the baud rate.

Let's try running `mcrun` in verbose mode. That will show the COM port it is trying to use. Just add `-v` to your command line:

```
mcrun -d -m -p esp32 -v
```

On my Mac, that outputs:

```
open -a /Users/xxx/Projects/moddable/build/bin/mac/release/xsbug.app -g
/Users/xxx/Projects/moddable/build/bin/mac/release/serial2xsbug /dev/cu.SLAB_USBtoUART 460800 8N1 -install /Users/xxx/Projects/moddable/build/bin/esp32/debug/helloworld/helloworld.xsa
```

What do you see?

ariznaf, 2020-11-10:
This is the output

```
d:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -p esp32 -v
# xsc example.xsb
        d:\users\fernando\ProyectosProg\moddable\build\bin\win\debug\xsc d:\Users\fernando\ProyectosProg\iot-product-dev-book\ch1-gettingstarted\helloworld\example.js -d -c -e -o d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\helloworld\files -r example
"# xsl "helloworld".xsa"
        d:\users\fernando\ProyectosProg\moddable\build\bin\win\debug\xsl -a -b d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\helloworld\files -n tech.moddable.helloworld -o d:\users\fernando\ProyectosProg\moddable\build\bin\esp32\debug\helloworld -r helloworld -u /   d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\helloworld\files\example.xsb d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\helloworld\files\check.xsb d:\users\fernando\ProyectosProg\moddable\build\tmp\esp32\debug\helloworld\files\mod\config.xsb
        tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
        tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start d:\users\fernando\ProyectosProg\moddable\build\bin\win\debug\xsbug.exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
        d:\users\fernando\ProyectosProg\moddable\build\bin\win\debug\serial2xsbug COM7 460800 8N1 -install d:\users\fernando\ProyectosProg\moddable\build\bin\esp32\debug\helloworld\helloworld.xsa

```

phoddie, 2020-11-10:
The trace from `mcrun` looks good.

Is it possible you have another console/terminal window open that is already connected to xsbug? If so, you need to close that first. (We usually run `mcconfig` and `mcrun` in the same console/terminal window so that doesn't happen)

ariznaf, 2020-11-10:
No, I am running all the commands from the same x86 native Tools command prompt for VS2019 (the console of visual studio 2019).

I have tried to close xsbug from mcconfig to mcrun or keep it opened, with same results.

phoddie, 2020-11-10:
OK. The screen shot above shows "main" in a tab. That means xsbug is connected to the device. Does the main tab disappear when you are at the command prompt, and appear after you execute mcrun? 

If you tap on main, you should see the Instruments pane on the left side of the window. If all is well, that should be updating once per second. What do you see there?

ariznaf, 2020-11-10:
This is what I can see if I tap in the main tab, after running mcrun
[image]

And if I press Ctrl+C in the command prompt, the main tab disappears

[image]

phoddie, 2020-11-10:
OK, that's really helpful. It means that there isn't a problem with the communication. The output suggests that the example wasn't installed. The problem must be with either the build of the example or the installation of the example.

Since you had a problem building the example the first time because of a manifest problem, I'm wondering if maybe the example binary is bad somehow. You could try forcing that to rebuild by deleting the object directory. That directory should be at [YOUR MODDABLE SDK ROOT]\⁨build⁩\⁨tmp⁩\⁨esp32⁩\⁨debug⁩\⁨helloworld⁩. After deleting the directory, execute `mcrun` again.

ariznaf, 2020-11-10:
Let me do the full cycle:

after running mcconfig in the host dir

[image]

Then I use ctrl+C in the console window
and I get this screen in xsbug
[image]

Then I run mcrun in the helloworld directory, in the same console (and without closing xsbug) and get this screen in xsbug
[image]

And after Ctrl+c I get a blank screen again
[image]

ariznaf, 2020-11-10:
> OK, that's really helpful. It means that there isn't a problem with the communication. The output suggests that the example wasn't installed. The problem must be with either the build of the example [...]

I have deleted the complete build/tmp dir and the bin/esp32 dir (it seem that it creates there the resulting apps too).

After that it seems that now it words.

Thank you very much for your time and help.

I think we can close the issue now.

[image]

phoddie, 2020-11-10:
Wahoo!! Congratulations. Thank you for your patience.

ariznaf, 2020-11-10:
The other answer was previous to your solution.

Now it works OK, thanks a lot, you have been the patient guy.

I close the issue.

closed 2020-11-10 by ariznaf