Moddable-OpenSource/iot-product-dev-book #5: [Question] ELF error while flashing host of chapter 3

ariznaf, 2020-11-13:
I am trying to test wifi connection and mqtt of chapter 3.
I am using a esp32 devkit C board.
It works with ch1 examples and ch6 examples.

But when I do mcconfig -d -m -p 32 to flash the host of the chapter I get at the end (when tho esp32 reboots) a n ELR file error is thown and it enter gdb.

If I reflash it with host in ch1 it works...

```
Leaving...
Hard resetting via RTS pin...
Done
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
ELF file SHA256: 615b3f8e0480ffd4731229f086ba0a253ef9c9d1ac22353e18b9651e44f3a275
Backtrace: 0x400887fc:0x3ffbee90 0x40088a3d:0x3ffbeeb0 0x400d2983:0x3ffbeed0 0x40144893:0x3ffbef00 0x4013a4d9:0x3ffbef30 0x4012a5e8:0x3ffbefa0 0x40113bf7:0x3ffbf140 0x40108510:0x3ffbf1b0 0x40108059:0x3ffbf250 0x401081b8:0x3ffbf2e0 0x40133b19:0x3ffbf370 0x40113e09:0x3ffbf410 0x400e2295:0x3ffbf480 0x400e068e:0x3ffbf4a0 0x400e06eb:0x3ffbf530 0x400e0f01:0x3ffbf550 0x400d2031:0x3ffbf700 0x400d203b:0x3ffbf750 0x40088d2e:0x3ffbf770
Entering gdb stub now.
```

phoddie, 2020-11-13:
There's nothing particularly unique about the Chapter 3 host. Since the Chapter 1 and Chapter 6 hosts work for you, we would expect Chapter 3 to work as well.

First, let's try a clean build, just in case. You can remove the build artifacts for the current project with:

```
mcconfig -d -m -p esp32 -t clean
```

Then, re-build as above:

```
mcconfig -d -m -p esp32 ssid="mywifiSSIDhere" password="*********"
```

Next, Chapter 3 does use Wi-Fi, while 1 and 6 do not. You might try running the Chapter 3 host without the Wi-Fi configuration settings on the command line, just to see if enabling Wi-Fi is somehow causing a problem with your board.

```
mcconfig -d -m -p esp32
```

ariznaf, 2020-11-13:
OK, I will try (my be I needed to use the -t clean before).

I need to connect to the wifi in order to test mqtt, that is why I was trying to use the command line connection (while I discover how to do it by programming it in the mqtt example).

phoddie, 2020-11-13:
Yes, I understand that you need to connect to Wi-Fi to use MQTT. I made the suggestion to disable Wi-Fi try to isolate the crash that you reported.

lprader, 2020-11-13:
I'm going to close this issue since it was resolved on our Gitter chatroom, but for anyone who may come across this, here are the steps to take to fix it:

1. Make sure the book repository is up to date

2. Erase the flash on your device using `esptool` (you'll have to change the path to the `esptool` directory to match your own setup):

```
cd /Users/lprader/esp32/esptool
esptool.py --port /dev/cu.SLAB_USBtoUART --chip esp32 erase_flash
```

3. Erase old builds of the host (you'll have to change the path to the `host` directory match your own setup):

```
cd /Users/lprader/iot-product-dev-book/ch3-network/host
mcconfig -d -m -p esp32 -t clean
```

4. Rebuild the host:

```
mcconfig -d -m -p esp32
```

5. Built the mqtt example:

```
cd ../mqtt
mcrun -d -m -p esp32 ssid=SSID_HERE password=PASSWORD_HERE
```

closed 2020-11-13 by lprader