Moddable-OpenSource/iot-product-dev-book #6: [Question] Is there any official forum about moddable and esp microcontroller development?

ariznaf, 2020-11-13:
I could not find a forum about moddable to get info and ask questions.

Is there any?

lprader, 2020-11-13:
There are several options to get info and ask questions.

- You can talk to us in our [Gitter chatroom](https://gitter.im/embedded-javascript/moddable)
- We respond to [issues in the Moddable SDK repository](https://github.com/Moddable-OpenSource/moddable/issues)
- You can reach out to us on [Twitter (@moddabletech)](https://twitter.com/moddabletech/)

ariznaf, 2020-11-13:
Ok, thank you. I am quite new to development under esp32 and javascript (I have some experience years ago wiht microcontrollers and C).
I did not want to overburden developers with my questions, that is why I was seeking for a user forum.

Moddable documentation is not too deep for a noob user.

lprader, 2020-11-13:
> I did not want to overburden developers with my questions, that is why I was seeking for a user forum.

Questions are no problem, we're happy to help! :)

closed 2020-11-13 by lprader