Moddable-OpenSource/iot-product-dev-book #8: [Bug] fatal error U1050

johnwargo, 2020-12-24:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

**Build environment:** macOS, Windows, or Linux

Windows

**Target device:** NodeMCU ESP8266, NodeMCU ESP32, Moddable One, Moddable Two, desktop simulator, etc.

Moddable Two

**Description**

A brief summary of the problem.

I've followed the development environment setup and have everything setup (I think) but when I try to deploy the host to the device, I get the following error:

```text
C:\Users\john\dev\iot-product-dev-book\ch1-gettingstarted\host>mcconfig -d -m -p esp32
D:\dev\hardware\moddable\moddable\build\tmp\esp32\debug\ch1host\makefile(167) : fatal error U1050: Detected ESP-IDF version 99b94d6. Expected ESP-IDF version v3.3.2.
```

When I run the same command for an `esp` device, it works.

I'm a little confused about the setup steps because you have me installing the Espressif IDF but also cloning it. Both instances end up on the Windows PATH. 

Looking at the cloned repo, I think I'm at the right branch, but its not working:

```shell
D:\dev\hardware\moddable\esp32\esp-idf>git checkout v3.3.2
Updating files: 100% (8670/8670), done.
Note: switching to 'v3.3.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 9e70825d1 Merge branch 'bugfix/sntp_init_can_run_before_net_connection_v3.3' into 'release/v3.3'

D:\dev\hardware\moddable\esp32\esp-idf>git submodule update

D:\dev\hardware\moddable\esp32\esp-idf>git status
HEAD detached at v3.3.2
nothing to commit, working tree clean

D:\dev\hardware\moddable\esp32\esp-idf>
```

I have 30 years of experience as a professional software developer and I've written 7 books on software development (6 on mobile development) topics. I'm only telling you this to save some time. 

If you are reporting a problem with the text, please include the page number.

If you are reporting a problem with an example, please include the name of the example.

If you are reporting a problem with a figure, please include the figure number (e.g. 6-12).

**Steps to Reproduce**

1. Build and install the host using this build command: `mcconfig -d -m ...`
2. Build and install the example using this build command: `mcrun -d -m ...`
2. Click on '....'
3. See error

**Images**

Screenshots or photos to help explain the problem

**Other information**

Any other details you believe may be helpful in understanding the issue, reproducing the unexpected behavior, or fixing the example

phoddie, 2020-12-24:
@johnwargo  - thanks for the report. We'll take a look and get back to you. It may take a couple days because of the holiday.

phoddie, 2020-12-24:
> I'm a little confused about the setup steps because you have me installing the Espressif IDF but also cloning it. Both instances end up on the Windows PATH.

I'm a little confused too. ;) The ESP32 set-up instructions for the ESP IDF [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#installing-1) have you do the following in Step 5:

```
cd ~/esp32
git clone -b v3.3.2 --recursive https://github.com/espressif/esp-idf.git
```

I just tried that and it works as expected.

I don't see a reference to "installing the Espressif IDF", only this `clone`. Apologies if I'm overlooking the install step. Would you provide the reference?

FWIW - the issue may be the `IDF_PATH` environment variable. The  Moddable SDK uses that for the ESP IDF if it is defined. I My guess is you have it defined but pointing to a version of the ESP IDF that is not version 3.3.2.  If `IDF_PATH` is not defined, then the Moddable SDK build uses a [default](https://github.com/Moddable-OpenSource/moddable/blob/9ed551497e76fc79f3e901d5a73fdd8f509233c3/tools/mcconfig/nmake.esp32.mk#L66-L69).

andycarle, 2020-12-24:
Hi @johnwargo,

Just to expand a bit on @phoddie's good advice above: 
 - You will need to have the `IDF_PATH` environment variable set on your system, as the default will not be correct for your paths (based on the console log you provided). In your case, that should point to `D:\dev\hardware\moddable\esp32\esp-idf`
 - You can verify your ESP-IDF clone and version at the command line by running `git describe --always` in your `%IDF_PATH%` directory.  On my system, that looks like this:

```console
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd %IDF_PATH%

C:\Users\andyc\esp32\esp-idf>git describe --always
v3.3.2
```

johnwargo, 2020-12-24:
> I'm a little confused too. ;) The ESP32 set-up instructions for the ESP IDF [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#installing-1) have you [...]

Step 2 [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md) says:

```text
Download and run the Espressif ESP-IDF Tools Installer. This will install the ESP32 Xtensa gcc toolchain, Ninja Build, and a Windows-based kconfig tool. This tool will also set your PATH to include the newly downloaded tools, as necessary.

It is safe to accept all of the default options in the installer, or to change install locations as necessary.

If you do not already have CMake or Python 2.7, the installer will also prompt you to download and install those tools (you should do so if needed).
```

Then step 5 on the same doc says to clone the Espressif IDF repo. 

Running that command delivers the same results, so why am I getting the error? Any ideas?

```text
D:\dev\hardware\moddable\esp32\esp-idf>git describe --always
v3.3.2
```

johnwargo, 2020-12-24:
> Hi @johnwargo,
> 
> Just to expand a bit on @phoddie's good advice above:
> 
> * You will need to have the `IDF_PATH` environment variable set on your system, as the default will not be correct for your paths (based on the console log you provided). In your case, that should point to `D:\dev\hardware\moddable\esp32\esp-idf`
> * You can verify your ESP-IDF clone and version at the command line by running `git describe --always` in your `%IDF_PATH%` directory.  On my system, that looks like this:
> 
> ```
> C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd %IDF_PATH%
> 
> C:\Users\andyc\esp32\esp-idf>git describe --always
> v3.3.2
> ```

I said I followed the instructions, so the `IDF_PATH` variable is already set and it points to the clone.

```text
IDF_PATH=D:\dev\hardware\moddable\esp32\esp-idf
```

phoddie, 2020-12-24:
> I'm a little confused about the setup steps because you have me installing the Espressif IDF but also cloning it

Perhaps you have assumed that  the "Espressif ESP-IDF Tools installer" does more than it actual does.

The "Espressif ESP-IDF Tools installer" installs the tools, as noted in the text you quoted. That includes the gcc toolchain, for example

The "Espressif ESP-IDF Tools installer" does not include the ESP IDF source code. That is retrieved using the `git clone` command.

andycarle, 2020-12-24:
@johnwargo  Curious!

I do see one hint in your build output. The error that you're getting (`fatal error U1050: Detected ESP-IDF version 99b94d6. Expected ESP-IDF version v3.3.2.`) indicates that it is seeing Git Commit 99b94d6. That is the current top-of-tree commit in the [iot-product-dev-book](https://github.com/Moddable-OpenSource/iot-product-dev-book) repository.

The part of the makefile that pulls the git repo version to check looks like this:
```
cd $(IDF_PATH) && git describe --always > $(TMP_DIR)\_idf_version.tmp 2> nul
```

I have to think that something strange is happening with your `IDF_PATH` variable still that is causing the `git describe --always` to run in your current working directory rather than in the ESP-IDF directory.

Since you've set that variable as indicated in the instructions, we have to assume that it is being overwritten somewhere. In the Command Prompt instance that you're using for the build, can you just do:
```console
echo %IDF_PATH%
```
and make sure that the output is what you think it should be?

If that looks sane, then the next place to check is in the makefile. You can do that by editing [`%MODDABLE%\tools\mcconfig\nmake.esp32.mk`](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/nmake.esp32.mk). Please add these two lines at Line 73 (just before the ESP-IDF version check):

```
!MESSAGE IDF_PATH: $(IDF_PATH)
!ERROR stop 
```

That will print out the `IDF_PATH` environment variable and then exit so that you can check it.

Please give those a shot and let me know what you see.

 &nbsp;- Andy

johnwargo, 2020-12-27:
> Perhaps you have assumed that the "Espressif ESP-IDF Tools installer" does more than it actual does.

you're right, I assumed the tooling included the SDK.

johnwargo, 2020-12-27:
> and make sure that the output is what you think it should be?

@andycarle it is, looks like this:

```shell
C:\Users\john>echo %idf_path%
D:\dev\hardware\moddable\esp32\esp-idf
```
which is exactly where I expect it to point.   Nothing's overwriting it. 

Just for grins, I whacked the clone and recreated a clean clone, then executed the following:

```shell
git checkout v3.3.2
git submodule update
```
and when I check my branch, I get the following:

```shell
D:\dev\hardware\moddable\esp32\esp-idf>git branch
* (HEAD detached at v3.3.2)
  master
```

johnwargo, 2020-12-28:
Got it working on macOS this morning but I'd still like to figure out how to get it to work on Windows as that's my primary development system. Will try out @andycarle's change today.

johnwargo, 2020-12-31:
@andycarle I don't have an ESP_IDF version check at line 73; looks like it's at line 47?  Are we looking at the same file?

[image]

johnwargo, 2020-12-31:
aaah, grabbed the wrong file. Sorry.

johnwargo, 2020-12-31:
@andycarle IDF Path set as expected, see figure

[image]

andycarle, 2021-01-06:
Hi @johnwargo, I hope you had a nice holiday! 

I must admit, this one has me a bit baffled. We can see that the path is right... and yet nmake does not seem to be changing to that directory before running the `git describe --always` bit. How odd.

The next logical step, I think, is to try simply deleting the ESP-IDF version check from the makefile fragment. Doing so will be harmless in this case. But, my guess is that whatever is causing the check to fail will also cause problems further along in the build.

You can remove lines 74-84 in [nmake.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/nmake.esp32.mk) and give it a shot... Perhaps the next place that it fails will provide some additional clues.

&nbsp;- Andy

johnwargo, 2021-01-06:
@andycarle OK, thanks, I'll give that a try.

johnwargo, 2021-01-06:
@andycarle here ya go! 

```shell
C:\Users\john\dev\book_code\iot-product-dev-book>cd ch10-piu

C:\Users\john\dev\book_code\iot-product-dev-book\ch10-piu>cd host

C:\Users\john\dev\book_code\iot-product-dev-book\ch10-piu\host>mcconfig -d -m -p esp_32/moddable_two
### Error: unknown platform: MAKE_FRAGMENT not found!

C:\Users\john\dev\book_code\iot-product-dev-book\ch10-piu\host>
```

andycarle, 2021-01-06:
@johnwargo That one is just a stray underscore in your mcconfig command.

It's `-p esp32/moddable_two`, not `-p esp_32/moddable_two`.

&nbsp;- Andy

phoddie, 2021-01-06:
A full list of Moddable SDK platform identifiers for ESP32 is [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#platforms), FWIW.

johnwargo, 2021-01-07:
@andycarle yep, absolutely - my bad. let me try it again with the right command line.

johnwargo, 2021-01-07:
Made it a lot further, but not all the way...

```shell
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.3
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************

C:\Users\john\dev\book_code\iot-product-dev-book\ch1-gettingstarted\host>mcconfig -d -m -p esp32/moddable_two
        1 file(s) copied.
Reconfiguring ESP-IDF...
Checking Python dependencies...
Python requirements from D:\dev\hardware\moddable\esp32\esp-idf\requirements.txt are satisfied.
Running cmake in directory D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=D:/dev/hardware/moddable/moddable/build/devices/esp32/xsProj/sdkconfig.mc -DSDKCONFIG_H=D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\idf\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\xsProj"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.30.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx asio jsmn aws_iot bootloader nimble bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: D:/dev/hardware/moddable/esp32/esp-idf/components/soc;D:/dev/hardware/moddable/esp32/esp-idf/components/log;D:/dev/hardware/moddable/esp32/esp-idf/components/heap;D:/dev/hardware/moddable/esp32/esp-idf/components/xtensa-debug-module;D:/dev/hardware/moddable/esp32/esp-idf/components/app_trace;D:/dev/hardware/moddable/esp32/esp-idf/components/freertos;D:/dev/hardware/moddable/esp32/esp-idf/components/vfs;D:/dev/hardware/moddable/esp32/esp-idf/components/newlib;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_ringbuf;D:/dev/hardware/moddable/esp32/esp-idf/components/driver;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_event;D:/dev/hardware/moddable/esp32/esp-idf/components/ethernet;D:/dev/hardware/moddable/esp32/esp-idf/components/mbedtls;D:/dev/hardware/moddable/esp32/esp-idf/components/micro-ecc;D:/dev/hardware/moddable/esp32/esp-idf/components/efuse;D:/dev/hardware/moddable/esp32/esp-idf/components/bootloader_support;D:/dev/hardware/moddable/esp32/esp-idf/components/partition_table;D:/dev/hardware/moddable/esp32/esp-idf/components/app_update;D:/dev/hardware/moddable/esp32/esp-idf/components/spi_flash;D:/dev/hardware/moddable/esp32/esp-idf/components/nvs_flash;D:/dev/hardware/moddable/esp32/esp-idf/components/lwip;D:/dev/hardware/moddable/esp32/esp-idf/components/tcpip_adapter;D:/dev/hardware/moddable/esp32/esp-idf/components/pthread;D:/dev/hardware/moddable/esp32/esp-idf/components/smartconfig_ack;D:/dev/hardware/moddable/esp32/esp-idf/components/wpa_supplicant;D:/dev/hardware/moddable/esp32/esp-idf/components/espcoredump;D:/dev/hardware/moddable/esp32/esp-idf/components/esp32;D:/dev/hardware/moddable/esp32/esp-idf/components/cxx;D:/dev/hardware/moddable/esp32/esp-idf/components/asio;D:/dev/hardware/moddable/esp32/esp-idf/components/jsmn;D:/dev/hardware/moddable/esp32/esp-idf/components/aws_iot;D:/dev/hardware/moddable/esp32/esp-idf/components/bootloader;D:/dev/hardware/moddable/esp32/esp-idf/components/nimble;D:/dev/hardware/moddable/esp32/esp-idf/components/bt;D:/dev/hardware/moddable/esp32/esp-idf/components/coap;D:/dev/hardware/moddable/esp32/esp-idf/components/console;D:/dev/hardware/moddable/esp32/esp-idf/components/nghttp;D:/dev/hardware/moddable/esp32/esp-idf/components/esp-tls;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_adc_cal;D:/dev/hardware/moddable/esp32/esp-idf/components/tcp_transport;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_http_client;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_http_server;D:/dev/hardware/moddable/esp32/esp-idf/components/esp_https_ota;D:/dev/hardware/mod [... cut ...]
D:\dev\hardware\moddable\moddable\build\tmp\esp32\moddable_two\debug\ch1host\mc.xs.c:4299:135: error: 'fx_DataView_prototype_setUint8Clamped' undeclared here (not in a function)
 /* 4006 */ { (txSlot*)&gxHeap[4007], {.ID = xsID_setUint8Clamped, .flag = 0x8f, .kind = XS_CALLBACK_X_KIND}, .value = { .callback = { fx_DataView_prototype_setUint8Clamped, NULL } }},
                                                                                                                                       ^
NMAKE : fatal error U1077: '"C:\Program Files\Espressif\ESP-IDF Tools\tools\bin\xtensa-esp32-elf-gcc.EXE"' : return code '0x1'
Stop.
```

andycarle, 2021-01-07:
Some progress, but still not into the ESP-IDF build.

The error you're seeing here is due to an incompatibility with an older version of the Moddable SDK tools binaries. Updates to the Moddable SDK occasionally require rebuilding the tools from scratch due to major changes in the XS JavaScript engine or to the tools themselves. When that happens, we note it in the release notes associated with that open source push. In this case, the breaking changes you're seeing [were noted here](https://github.com/Moddable-OpenSource/moddable/releases/tag/OS201230).

To fix things up, you just need to delete your `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories. Then rebuild the Moddable SDK tools. 

The whole process is:
```console
C:\Users\andyc>cd %MODDABLE%\build

C:\Users\andyc\Projects\moddable\build>rmdir /s /q bin

C:\Users\andyc\Projects\moddable\build>rmdir /s /q tmp

C:\Users\andyc\Projects\moddable\build>cd makefiles\win\

C:\Users\andyc\Projects\moddable\build\makefiles\win>build
```

&nbsp;- Andy

johnwargo, 2021-01-13:
@andycarle thanks, that worked. I'm up and running on Windows. Thanks again.

closed 2021-01-13 by phoddie

andycarle, 2021-02-26:
@johnwargo I know this thread is now ancient, but I finally this evening realized what was going on with the ESP-IDF version check on your build machine. 

The issue is that you are building on a different drive than the one that holds the ESP-IDF, which our git version check did not anticipate. Because you were building on the C drive but your `%IDF_PATH%` is on the D drive, the `cd $(IDF_PATH)` isn't doing what we want: it's just changing the working directory on the D drive without actually moving into it. 

The fix is to change the [nmake.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcconfig/nmake.esp32.mk) line from 
```make
!IF [cd $(IDF_PATH) && git describe --always --abbrev=0 > $(TMP_DIR)\_idf_version.tmp 2> nul] == 0` 
```
to 
```make
!IF [cd /D $(IDF_PATH) && git describe --always --abbrev=0 > $(TMP_DIR)\_idf_version.tmp 2> nul] == 0
```

I've made that change to the Moddable SDK locally and it will be reflected in our next push to GitHub.

&nbsp;- Andy

johnwargo, 2021-02-26:
@andycarle Thanks so much for figuring this out and letting me know. I'll give it a shot with the next SDK release.