Moddable-OpenSource/iot-product-dev-book #9: MOD examples don't work

patxirb, 2021-01-03:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

**Build environment:** macOS, Windows, or Linux

**Target device:** NodeMCU ESP8266, NodeMCU ESP32, Moddable One, Moddable Two, desktop simulator, etc.

**Description**

A brief summary of the problem.

If you are reporting a problem with the text, please include the page number.

If you are reporting a problem with an example, please include the name of the example.

If you are reporting a problem with a figure, please include the figure number (e.g. 6-12).

**Steps to Reproduce**

1. Build and install the host using this build command: `mcconfig -d -m ...`
2. Build and install the example using this build command: `mcrun -d -m ...`
2. Click on '....'
3. See error

**Images**

Screenshots or photos to help explain the problem

**Other information**

Any other details you believe may be helpful in understanding the issue, reproducing the unexpected behavior, or fixing the example

lprader, 2021-01-04:
Hello, sorry you're having trouble!

To clarify, are you able to install the host for chapter 3? The section **Installing the Network Host** at the top of page 137 doesn't explicitly state the build command, but it's the same pattern as described in chapter 1 (that you say you were able to use):

```
cd $EXAMPLES/ch3-network/host
mcconfig -d -m -p esp
```

After you install the host, you should be able to install examples with mcrun. The build command you showed looks correct for that step.

patxirb, 2021-01-04:
Hello,

cd $EXAMPLES/ch3-network/host
mcconfig -d -m -p esp32 ssid="WifiNi" password="miPasswo0rd"

Yes, yes. All this works perfectly

[image]

But not this:

cd $EXAMPLES/ch3-network/http-get
mcrun -d -m -p esp32

[image]

I translate for you:

Se agotó el tiempo del semáforo => Semaphore timed out.

All my attempts come here. I cannot advance at this point.

lprader, 2021-01-05:
The problem may be that your `UPLOAD_PORT` environment variable isn't set correctly; it should be set to the COM port for your device. 

Step 8 of our [ESP32 installation instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#installing-1) for Windows explains how to do that, but it's described as optional for advanced users. I think it may be necessary for anyone using `mcrun`, which of course includes book readers :)

Here are the steps:

1. Plug your ESP32 into your computer
2. Launch the Windows Device Manager
3. Open the "Ports (COM & LPT)" section
4. Look for "Silicon Labs CP210x USB to UART Bridge" and note the associated COM port (e.g. `COM3`)
5. Open the "Environment Variables" dialog of the Control Panel app by following [these instructions](https://www.architectryan.com/2018/08/31/how-to-change-environment-variables-on-windows-10/)
6. From that dialog, create a User Variable called `UPLOAD_PORT` and set it to the COM port you noted earlier.
7. Newly-set environment variables will not take effect in existing Command Prompt instances, so be sure to open a new Command Prompt instance after applying these changes

Give that a try and let me know if it fixes the problem.

patxirb, 2021-01-05:
Looking for the solution, it is seen that I finally left the configuration wrong.

I have reinstalled everything from scratch and the symptoms have changed, but the problem persists.

During the careful reinstallation I discovered a bug in the program. In my case I have

HOMEDRIVE=D:

and

HOMEPATH=\Users\Patxi

but the installation uses

HOMEDRIVE=C:

and

HOMEPATH=\Users\Patxi

So in the variable
PATH
create the entry:

C:\user\Patxi\esp32\esp-idf\tools

Instead of:

D:\Users\Patxi\esp32\esp-idf\tools

But after making the correction the initial problem remains.

I have made a video with a concrete example. But the same thing happens with any of the MODs. Surely it is some detail that I cannot detect, but I have spent many hours trying to solve it without success.

https://user-images.githubusercontent.com/32263158/103698273-67848880-4fa1-11eb-9de9-aaf496dbc822.mp4

I really appreciate your dedication and help,

lprader, 2021-01-06:
> I really appreciate your dedication and help,
I'm happy to help! And I appreciate your patience.

The issue in the video you shared looks very different than the initial problem. The initial problem was a build error that seems to be resolved now that you've fixed your environment variables.

The issue I see in the video is that the host is not finding the mod after you install it. Since the host doesn't find the mod, it doesn't bother trying to connect to Wi-Fi (which is the expected behavior).

Can you check that your copy of this repository and the Moddable SDK are up to date?

The most recent commit in this repository should be this one:

```
lprader:iot-product-dev-book lprader$ git log
commit 99b94d6053e5c424e5335281eae298d4f690823c (HEAD -> master, origin/master, origin/HEAD)
Author: Peter Hoddie <peter@moddable.tech>
Date:   Fri Dec 18 10:04:11 2020 -0800

    convert function unnecessary now that XS handles undefined values in template literals
```

And the most recent commit in your copy of the Moddable SDK repository should be this:

```
lprader:moddable lprader$ git log
commit 86a98025fe3ea5d4c658e87c7ab9d368db60c4a8 (HEAD -> public, origin/public, origin/HEAD)
Author: Peter Hoddie <peter@moddable.tech>
Date:   Tue Jan 5 14:10:00 2021 -0800

    add mcrun, small updates
```

patxirb, 2021-01-07:
> I'm happy to help! And I appreciate your patience.

@lprader you are the winner of the courtesy duel. I've get out of words.

I had the Moddable SDK installed with commit;

`D:\Usuarios\Patxi\Projects\moddable>git log

commit 4ad0dde1b7d294b59d15a07e7660a14415ff66e5 (HEAD -> public, origin/public,origin/HEAD)
Author: Michael Kellner <mkellner@moddable.tech>
Date:   Thu Dec 31 18:16:55 2020 -0800`

I have updated it and now it is with the same one that you indicate

`D:\Usuarios\Patxi\Projects\moddable>git log
commit 86a98025fe3ea5d4c658e87c7ab9d368db60c4a8 (HEAD -> public, origin/public,origin/HEAD)
Author: Peter Hoddie <peter@moddable.tech>
Date:   Tue Jan 5 14:10:00 2021 -0800

    add mcrun, small updates`

**And now EVERYTHING works perfectly.**

I don't have words but I do have one word:

**gratefulness**

lprader, 2021-01-08:
Excellent. I'll go ahead and close this issue.

closed 2021-01-08 by lprader