Moddable-OpenSource/iot-product-dev-book #10: [Bug] mcrun M5Stack Core2 ready?

0xABCDE, 2021-01-26:
Hi there,

Setup:

> Windows 10 x64
> Moddable SDK `esp-idf-4` branch
> ESP-IDF 4.2 configured as per instruction in wiki

I am following the example on this book using and M5Stack Core device. running `mcconfig` on any example with the following command works fine, for example:

`> cd %EXAMPLES%\ch1-gettingstarted\helloworld-gui`
 `> mcconfig -d -m -p esp32/m5stack_core2 `

However, installing examples via `mcrun` is currently not working even though installing the host works fine. Example:

> installing host (works fine)
`> cd %EXAMPLES%\ch3-network\host`
`> mcconfig -d -m -p esp32/m5stack_core2 `

> installing example (giving errors)
`> cd %EXAMPLES%\ch3-network\http-get`
`> mcrun -d -m -p esp32/m5stack_core2 `

setting `VERBOSE=1`, few problems were observed:

1. UPLOAD_PORT environment variable is not detected, when `mcrun` command is used, causing it to default to COM3 and default debugger speed
2. running serial2xsbug with the `-install` argument directly on the cmd, the example *seems* to be uploading fine, but no changes whatsover can be observed on the xsbug or core2 screen, except that it seems like the host code is just being refreshed.

Any idea regarding this? Thanks

phoddie, 2021-01-26:
Sorry for the trouble. It may bee that core2 serial connection only operates at a non-standard baud rate. Some other recent m5atom devices are like that. Since you are already invoking serial2xsbug directly from the command line, try setting the baud rate to 1500000.

0xABCDE, 2021-01-26:
Hi @phoddie thank you for the (blazing) fast reply. I tried invoking serial2xsbug directly like this with 1500000 baud:

`C:\Users\K\Projects\moddable\build\bin\win\release\serial2xsbug COM4 15000000 8N1 -install C:\Users\K\Projects\moddable\build\bin\esp32\debug\http-get\http-get.xsa`

gives the uploading sound, followed by nothing in xsbug. same behaviour

0xABCDE, 2021-01-26:
I also noticed that unlike when using `mcconfig`, the .xsa files get built into `..\build\bin\esp32\debug\http-get` folder instead of `..\build\bin\esp32\m5stack_core2\debug\http-get`. Not sure if that is relevant anyhow

phoddie, 2021-01-26:
**Note**: The correct baud rate is 1500000 (1.5 Mbps). Your command line appears to have an extra zero.

0xABCDE, 2021-01-26:
> **Note**: The correct baud rate is 1500000 (1.5 Mbps). Your command line appears to have an extra zero.

Apologies, a typo. I did use 1500000 as baud rate

andycarle, 2021-01-26:
Hi @0xABCDE,

Sorry to hear that you're having trouble!

A couple of ideas:

 1) Could you clarify what errors you're seeing when you use `mcrun` to do the deployment? I just want to make sure that there aren't any issues with the build itself; just the deployment. If you could post the full output from `mcrun` with `VERBOSE=1` set that would be helpful.
 2) I think @phoddie is likely on the right track with trying different baud rates. I would recommend trying `921600` -- that is what we have set for the debugger speed in the `m5stack_core2` manifest, but I suspect that value is not being used correctly in `mcrun`. So, try: `C:\Users\K\Projects\moddable\build\bin\win\release\serial2xsbug COM4 921600 8N1 -install C:\Users\K\Projects\moddable\build\bin\esp32\debug\http-get\http-get.xsa`

Once we get this working by running `serial2xsbug` directly, we'll go back and figure out what's wrong with mcrun.

&nbsp;- Andy

0xABCDE, 2021-01-26:
Hello @andycarle many thank for the reply. Here are the results:

1. the full erors are as follows with `VERBOSE=1`

> mcrun -d -m -p esp32/m5stack_core2

`tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
        tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start C:\Users\K\Projects\moddable\build\bin\win\release\xsbug.exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
        C:\Users\K\Projects\moddable\build\bin\win\release\serial2xsbug COM3 460800 8N1 -install C:\Users\K\Projects\moddable\build\bin\esp32\debug\http-get\http-get.xsa
C:\Users\K\Projects\moddable\tools\serial2xsbug\serial2xsbug_win.c(135): The system cannot find the file specified.
NMAKE : fatal error U1077: 'C:\Users\K\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0x1'
Stop.`

2. Tried with `921600` baud rate. Same behaviour, the program seems to rerun the host, not the mod.
`C:\Users\K\Projects\moddable\build\bin\win\release\serial2xsbug COM4 921600 8N1 -install C:\Users\K\Projects\moddable\build\bin\esp32\debug\http-get\http-get.xsa`

andycarle, 2021-01-26:
Thank you for the info. 

I've been able to reproduce the issue you're seeing on an M5GO. I'll check back in here when I've sorted out a fix.

0xABCDE, 2021-01-27:
Many thanks @andycarle

andycarle, 2021-01-27:
@0xABCDE After getting into a state where I was seeing the same thing you're seeing, I was able to get back to a working state by erasing the flash on my M5GO. Annoyingly, I have not been able to get back to the broken state again to diagnose further. 

I recommend erasing the flash on your device before proceeding to see if it helps. You can do that with this command:

```console
python %IDF_PATH%\components\esptool_py\esptool\esptool.py erase_flash
```

If your device's port is not automatically detected, you can specify it with the `-p port` option, like this:

```console
python %IDF_PATH%\components\esptool_py\esptool\esptool.py -p COM4 erase_flash
```

Please give that a shot, then trying installing the host with `mcconfig` and the mod with `mcrun`. 

Also: thank you for the report about `UPLOAD_PORT` not setting properly with `mcrun`. I have checked in a fix for that which will go live with our next open source push. But, for now, you can fix the issue locally by editing [tools/mcrun/nmake.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/esp-idf-4/tools/mcrun/nmake.esp32.mk). 

Please change the block that is currently:

```
!IF "$(UPLOAD_PORT)"==""
!IF [python $(PROJ_DIR_TEMPLATE)\getPort.py $(IDF_PATH)\tools > $(TMP_DIR)\_default_port.tmp 2> nul] == 0
DEFAULT_PORT = \
!INCLUDE $(TMP_DIR)\_default_port.tmp
!IF [del $(TMP_DIR)\_default_port.tmp] == 0
!ENDIF
!ENDIF
UPLOAD_PORT = $(DEFAULT_PORT)
!ELSE
UPLOAD_PORT = COM3
!ENDIF
```

to:

```
!IF "$(UPLOAD_PORT)"==""
!IF [python $(PROJ_DIR_TEMPLATE)\getPort.py $(IDF_PATH)\tools > $(TMP_DIR)\_default_port.tmp 2> nul] == 0
DEFAULT_PORT = \
!INCLUDE $(TMP_DIR)\_default_port.tmp
!IF [del $(TMP_DIR)\_default_port.tmp] == 0
!ENDIF
!ENDIF
UPLOAD_PORT = $(DEFAULT_PORT)
!ENDIF
```

Apologies for the inconvenience!

&nbsp;- Andy

0xABCDE, 2021-01-28:
Hi @andycarle thank you for getting back. 

I flashed the Core2 using

`python %IDF_PATH%\components\esptool_py\esptool\esptool.py -p COM4 erase_flash`

Output:

```
esptool.py v3.0
Serial port COM4
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:8f:18:38
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 23.0s
Hard resetting via RTS pin...
```

I guess the flash was successful? The device did not restart or anything. I then tried uploading the host via `mcconfig -d -m -p esp32/m5stack_core2` followed by the mod via `mcrun -d -m -p esp32/m5stack_core2`.

Same issue as before. The uploading sound is played, there was no error as well, however, the mod was not running.

Also, I can confirm the patch to nmake.esp32.mk fixed the UPLOAD_PORT error.

Thanks.

0xABCDE, 2021-01-28:
Hi @andycarle some update.

In Chapter 3 and 4, where the host install does not require a screen, I have managed to run the mod using `mcrun` successfully via the following method:

1. Install host using `mcconfig -d -m -p esp32`  (setting the platform to `esp32` instead of `esp32/m5stack_core2`
2. run the mod using `mcrun -d -m -p esp32/m5stack_core2`

For chapter 9 onwards, installing the host requires a screen to be configured thus the above command (1) no longer works. If I install the host using `mcconfig -d -m -p esp32/m5stack_core2` followed by `mcrun -d -m -p esp32/m5stack_core2`, previous problem persists, in which the mod doesn't run. 

I am starting to think this problem is specific to perhaps Core2?

Thanks

andycarle, 2021-01-28:
Interesting! 

Since we know the default `UPLOAD_SPEED` and `DEBUGGER_SPEED` for the ESP32 are working for you (because `-p esp32` is working for both `mcconfig` and `mcrun`), let's just try getting rid of those settings in the `m5stack_core2` manifest as a test.

Try this, if you could. In [%MODDABLE%\build\devices\esp32\targets\m5stack_core2\manifest.json](https://github.com/Moddable-OpenSource/moddable/blob/esp-idf-4/build/devices/esp32/targets/m5stack_core2/manifest.json), please delete these 4 lines:

```json
	"build": {
		"UPLOAD_SPEED": "1500000",
		"DEBUGGER_SPEED": "921600"
	},
```

Then rebuild/deploy the host for the esp32/m5stack_core2 platform (`mcconfig -d -m -p esp32/m5stack_core2`) and then install the mod for the same platform (`mcrun -d -m -p esp32/m5stack_core2`).

Let me know if that helps the situation. If it does, I have a pretty good idea of what the underlying issue is.

Thanks!
&nbsp;- Andy

0xABCDE, 2021-01-28:
Hi @andycarle. Edited the manifest and installed the Chapter 10 host and `multiple screen`  mod with no luck. I also tried deleting the existing build from `%MODDABLE%\build\bin\esp32\m5stack_core2\debug\` (deleting the ch10host folder) and rebuilding, but just like previous tries, upon running `mcrun` the screen refreshes with the text `Ready to install ...` and nothing else

andycarle, 2021-01-29:
Very curious, indeed. 

I'm thinking on a couple of other potential solutions. In the meantime, I've ordered an M5Stack Core2 for myself. All my M5 devices are of older vintage, so it was probably time for an update.

That should arrive over the weekend. I'll give it a shot when it gets here and see if I learn anything from using it.

0xABCDE, 2021-01-29:
Hi @andycarle thank you for you help with this and sorry for (any) trouble caused.

andycarle, 2021-02-02:
Hi @0xABCDE,

My M5Stack Core2 arrived and I was able to test with it today. I've confirmed the issue that you're seeing here when building with top-of-tree sources: I can deploy the host and it works correctly, but mcrun does not work as expected.

However, I was able to work around the problem by deleting the "build" section from the m5stack_core2 manifest, as described above. I think it would be worth trying that workaround again on your computer, just to see if it has the effect I would expect. 

To do that:
1) Delete the 4 lines of the "build"  section from [%MODDABLE%\build\devices\esp32\targets\m5stack_core2\manifest.json](https://github.com/Moddable-OpenSource/moddable/blob/esp-idf-4/build/devices/esp32/targets/m5stack_core2/manifest.json) as described above.
2) Completely delete all output files from the ESP32 build. Do this by completely deleting `%MODDABLE%\build\bin\esp32` and `%MODDABLE%\build\tmp\esp32`. (I suspect you may have had issues trying this before due to not getting the `tmp` directory.)
3) Rebuild and deploy the `ch3-network/host` app using `mcconfig -d -m -p esp32/m5stack_core2`.
4) Deploy the `http-get` mod using `mcrun -d -m -p esp32/m5stack_core2 ssid=networkName password=networkPassword`

Please give that a shot again, if you could, and let me know if the outcome is any different than before. If so, I have a pretty good idea of the fix to make in the Moddable SDK.

Thanks!
&nbsp;- Andy

0xABCDE, 2021-02-02:
Hello @andycarle. At last, I can confirm that this indeed works! Apparently deleting the `\esp32` in both `build` and `temp` and rebuilding makes the `mcrun` functioning as intended on core2. I can confirm that this also works with the `multiple-screen` example in chapter 10 (piu). 

Many thanks for your dedication!

phoddie, 2021-02-04:
Today's update to the Moddable SDK contains the changes discussed here. @andycarle has confirmed that mods install to M5Stack Core2 without any special steps. Tank you for your patience in working through this. I believe we can close this issue out now.

closed 2021-02-10 by phoddie