Moddable-OpenSource/iot-product-dev-book #11: [Bug] Touch mapping does not align with LCD Display when running examples

dbhaig, 2021-03-12:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

**Build environment: Linux

**Target device: Moddable Zero (i.e. ESP32 wired to an ILI9341 240x320 Touch TFT Module)

**Description**
When I run examples/piu/keyboard:  mcconfig -m -p esp32/moddable_zero

The touch sensing seem to be mirrored along the horizontal axis.  

 - to activate the space bar I have to touch the LCD near the top
 - touching the '#' key activates the '1' or '2', depending on where I touch it

The map-puzzle example exhibits similar misalignment of the image and touch response.

phoddie, 2021-03-12:
The touch screen needs to be calibrated for your device. On some boards the touch sensor is rotated. On many it is off center. It can also be a little bigger or smaller than the display.

The XPT2046 driver has [code to apply calibration](https://github.com/Moddable-OpenSource/moddable/blob/e7f318d3eaa45b24f7329ddf0305c096d4f31c4f/modules/drivers/xpt2046/modXpt2046.c#L268-L276) data stored in the device preferences.

To calibrate your device you just need to run another example. ;) The example is here: [$MODDABLE/examples/drivers/xpt2046calibrate](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/drivers/xpt2046calibrate/main.js).

dbhaig, 2021-03-13:
As you said, running xpt2046calibrate resolved the issue. Thanks.

closed 2021-03-13 by dbhaig