Moddable-OpenSource/iot-product-dev-book #12: Is an "ESP32 NodeMCU module" an ESP32-DevkitC?

mburr, 2021-08-04:
In the Hardware Requirement's section, a minimal setup for ESP32 is described as an "ESP32 NodeMCU module".  The best that I can figure from searching is that this is an early name for the commonly available ESP32-DevkitC of which there are several related designs:

  - https://www.espressif.com/en/products/devkits/esp32-devkitc

Should I expect to be successful using any of those DevkitC modules (or of the hundreds of clones)?  Are there models that are known to have problems (more than some IOs being routed to different pins)?

phoddie, 2021-08-04:
Yes, the NodeMCU should work with any of those ESP32-DevkitC boards (though I can't say we have tried every one of them). The `esp32/nodemcu` build target is pretty [bare bones](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/nodemcu/setup-target.js). It just defines classes for the flash button and LED. If those get in the way, you cam just build with the `esp32` platform target.

mburr, 2021-08-09:
Thanks - that's what I was hoping.  I had just never heard the term "NodeMCU" to describe those DevKitC boards before.  But looking around it seems it's used more than I realized.  So I'll be able to get started with what I already have on hand - thanks.

Closing the issue.

closed 2021-08-09 by mburr

eltoroit, 2021-12-27:
I am not familiar with microcontrollers,  I am a JavaScript developer learning electronics and I have done a bit of Raspberry PI.

I do not know what NodeMCU nor DevKitC is… 

Can I use any board that has a ESP32 chip? Or does it need anything special. Are there special type of ESP32 chips that make them NodeMCU or DevKitC?

thanks

phoddie, 2021-12-27:
The Moddable SDK supports many ESP32 boards. Our Getting Started guide features [several](https://github.com/Moddable-OpenSource/moddable#esp32-by-espressif) and links to a more [complete list](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#platforms) including our own Moddable Two.

eltoroit, 2021-12-27:
Thanks for the prompt reply, but let me ask it differently…

Are there any ESP32 that are not supported?

I know it’s impossible to test out every single board, but is there anything in the name that would indicate it’s not going to work? Should I be careful with ESP32 that indicate a specific feature?

I guess this is an advantage of the Raspberry PI, there are very few options, so it’s hard to go wrong ;-)

phoddie, 2021-12-27:
The ESP32 MCU is basically the same across all boards. In theory, anyone of them can work. In practice, there are differences in configuration, how they are programmed, and peripheral support that can be complex to solve and require deep understanding. If you are just getting started, it is best to start with a board that is already supported so you can build on the work others have already done to support it.

eltoroit, 2021-12-27:
Thanks for your answer, I will follow your advise