Moddable-OpenSource/iot-product-dev-book #14: Page 129 contains incorrect instructions wrt mcconfig/mcrun

JoshuaPK, 2022-02-13:
Page 129 of the book contains the notice: _This section uses a different installation pattern from the one you learned in Chapter 1: rather than installing the host with mcconfig and then installing the examples using mcrun, you install the examples using mcconfig._

This does not appear to be correct.  If you attempt to run mcconfig to deploy any of the network examples, you get an error about a missing modTimer.h file.  If, however, you run mcconfig to deploy the 'host' that lives in the directory for the network chapter, you can then deploy the examples themselves using 'mcrun', and it works.

lprader, 2022-02-13:
Ah, I see how that's confusing. But that sentence is only referring to the section **Connecting to Wi-Fi** (not the entire chapter), which covers the following examples:

- `$EXAMPLES/ch3-network/wifi-command-line`
- `$EXAMPLES/ch3-network/wifi-code`
- `$EXAMPLES/ch3-network/wifi-open-ap`

To install those three examples, you should use `mcconfig`. In subsequent sections, you should install the host with `mcconfig` and  install the examples with `mcrun` as described on page 137.

eltoroit, 2022-02-13:
I had the same confusion, maybe the note should have provided the name of the section ;-)

> The "Connecting to Wi-Fi" section uses a different installation pattern from the one you learned in Chapter 1: rather than installing the host with mcconfig and then installing the examples using mcrun, you install the examples using mcconfig.