Moddable-OpenSource/iot-product-dev-book #17: Issues getting any graphics examples to work

cdctx07, 2022-03-25:
Hi. I am trying to see the examples of CH9 and CH10 of the book. Normally, I can get any of the /Projects/moddable/examples to work, but when I go to install the book examples, I run into walls and errors.

So, for CH10, I will first run "mcconfig -d -m -p esp32/moddable_two" in the /host folder. It does its thing, and my Moddable Two reboots and shows "Read to install apps!" on the screen. Great!

I then go to /ch10-piu/multiple-screens and execute "mcrun -d -m -p esp32/moddable_two" and I either run into serial2xsbug.exe errors or my command box screen says "Starting xsbug..." and just sits there. It will launch xsbug and do absolutely nothing. Even if I remove "-d" and press enter, it just sits and does nothing.

I've now moved over to /ch10-piu/tiled-background and ran "mcrun -m -p esp32/moddable_two," which gives me the following then sits there without sign of anything happening:
# xsc example.xsb
# xsc check.xsb
# xsc mod\config.xsb
# png2bmp tile-color.bmp tile-alpha.bmp
"# xsl "tiled-background".xsa"

Again, I can install anything from /moddable/examples without any issue. It's getting anything from this book to run that's the problem.

Any advice on getting these examples to work after I mcconfig the host folders?

phoddie, 2022-03-25:
Thank you for the report and apologies for the trouble.

It looks like building and install the host with `mcconfig` works as expected. That's great.

It is the mod that is failing. I'll look into that later today. Meanwhile, two questions for you:

1. What kind of computer are you using to build? It looks like it might be Windows, but I'm not sure.
2. You note that the examples in Chapters 9 and 10 fail. Do the mods in the earlier chapters work for you?

phoddie, 2022-03-25:
We tried this using the current Moddable SDK on both macOS and Windows. Fortunate/unfortunately, it worked correctly on both.

You mentioned that sometimes you "run into serial2xsbug.exe errors". Would you share those error messages? Perhaps they offer a clue.

It isn't 100% clear if xsbug is working for you. When you run an example like `$MODDABLE/examples/helloworld`, do see traces in the xsbug console and breakpoints.

If you know the serial port name that your Moddable Two is connected to (look at the output of mcconfig or check Device Manager), you can try setting that explicitly when executing `mcrun` in case the auto-detection isn't working:

```console
SET UPLOAD_PORT=COM3
mcrun -d -m -p esp32/moddable_two
```

Some other ideas to try, just in case:
- Make sure you have [updated](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#table-of-contents) to the the latest Moddable SDK (and rebuild tools, as the documentation describes)
- Clear flash on the target device (`esptool.py erase_flash`)
- Make sure to exit `mcconfig` with Control-C after the host is up-and-running, before executing `mcrun` to set-up the mode

cdctx07, 2022-03-25:
Thanks @phoddie 

I am using Windows 11 on a Surface Book Pro. I am playing with Ch 1 stuff now. Host installed fine with mcconfig, with xsbug starting and showing me the ready for install message in the log.

I just ran mcrun -d -m -v -p esp32/moddable_two in the Ch 1 helloworld folder after setting my port to COM5 (it was already looking there, but just made sure once more). xsbug.exe has opened. I'm stuck here now:

-----
C:\Users\<name>\Projects\iot-product-dev-book\ch1-gettingstarted\helloworld>mcrun -d -m -v -p esp32/moddable_two
        tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
        tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start C:\Users\<name>\Projects\moddable\build\bin\win\release\xsbug.exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
Starting xsbug...
        C:\Users\<name>\Projects\moddable\build\bin\win\release\serial2xsbug COM5 460800 8N1 -install C:\Users\<name>\Projects\moddable\build\bin\esp32\debug\helloworld\helloworld.xsa
-----

I used -v so I could see what was happening in the command prompt besides "Installing mod." with nothing else. But it just sits here now, even if I leave it for a bit.

I'll let this sit here for a bit to see what happens. If nothing, I'll go back to Ch 9 and 10 stuff to see if I can recreate the serial2xsbug.exe errors.

cdctx07, 2022-03-25:
So my command prompt just sat as I had it above. Nothing happened after that.

I went back to Ch 10 and installed the host. I now have "Ready to install apps!" on the Moddable Two screen. So I went to tiled-background and an mcrun -d -m -v -p esp32/moddable_two. 

```console
C:\Users\<name>\Projects\iot-product-dev-book\ch10-piu\tiled-background>mcrun -d -m -v -p esp32/moddable_two
# xsc example.xsb
        C:\Users\<name>\Projects\moddable\build\bin\win\release\xsc C:\Users\<name>\Projects\iot-product-dev-book\ch10-piu\tiled-background\example.js -d -c -e -o C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files -r example
# xsc check.xsb
        C:\Users\<name>\Projects\moddable\build\bin\win\release\xsc C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\check.js -d -c -e -o C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files -r check
# xsc mod\config.xsb
        C:\Users\<name>\Projects\moddable\build\bin\win\release\xsc C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\mod.config.js -d -c -e -o C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\mod -r config
# png2bmp tile-color.bmp tile-alpha.bmp
        C:\Users\<name>\Projects\moddable\build\bin\win\release\png2bmp C:\Users\<name>\Projects\iot-product-dev-book\ch10-piu\tiled-background\tile.png -f rgb565le -o C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files -r 90 -n tile
"# xsl "tiled-background".xsa"
        C:\Users\<name>\Projects\moddable\build\bin\win\release\xsl -a -b C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files -n tech.moddable.tiled-background -o C:\Users\<name>\Projects\moddable\build\bin\esp32\debug\tiled-background -r tiled-background -u /   C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\example.xsb C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\check.xsb C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\mod\config.xsb  C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\tile-color.bmp C:\Users\<name>\Projects\moddable\build\tmp\esp32\debug\tiled-background\files\tile-alpha.bmp
        tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
        tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start C:\Users\<name>\Projects\moddable\build\bin\win\release\xsbug.exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
Starting xsbug...
        C:\Users\<name>\Projects\moddable\build\bin\win\release\serial2xsbug COM5 460800 8N1 -install C:\Users\<name>\Projects\moddable\build\bin\esp32\debug\tiled-background\tiled-background.xsa
Installing mod.
```

It's just sitting at mod. Xsbug opened, but nada is happening, and my screen still sits at "Ready to install apps!"

phoddie, 2022-03-25:
Thanks for the notes. That all looks reasonable. Let's try to coax some more information out of `serial2xsbug` (the tool that bridges between the serial connection on Moddable Two and xsbug). Edit [serial2xsbug.h](https://github.com/Moddable-OpenSource/moddable/blob/40cf4e46b2965b3a5a064b0b8da502750c87d238/tools/serial2xsbug/serial2xsbug.h#L91-L92) to turn on `mxTrace` and `mxTraceCommands`. 

```c
#define mxTrace 1
#define mxTraceCommands 1
```

Then rebuild Moddable SDK tools. And try `mcrun` again. When I do that, here's what I see (the garbage is because of a temporary baud rate mismatch when the ESP32 starts-up):

```console
origin$ mcrun -d -m -p esp32/moddable_two
### restart
### fxRestart
x?x?x?x??xxxx?x??????????????????????????x?????x?xxx??x?x?x????x??xxxxx?x??x?xx??xxxxxx?xx??x?x?x??x??????xxx?xx?x?xx??xx?xxx??xxx?xx?x??x?x?x??x?x?x??xx??????xxx?xxxx????xxxxx?x??xxxxx?x???x??xxx?x??x??xx?x???x????x??xx?x???x??xxx?x???x????xxxx?x???x??xxx?x?x?xxx???xx???x?xxxx??x??x?xxx?xx?x???x????x?x?x??????x?x?????????xx?xx?x??x???????????xxx?x??x????????x?xx?xx?x???x???????????x??????xx?x?x?x?x?x?x??????x???x?x<?xs-00000000?>

<?xs.3FFB5658?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB5658?>### install

<?xs#3FFB5658?>
<?xs#3FFB5658?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### installSpace: 262144, fragment size 4096
### install fragment @ 0 size 4

<?xs#3FFB5658?>
<?xs#3FFB5658?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### install fragment @ 4 size 1367

<?xs#3FFB5658?>^??
<?xs#3FFB5658?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### update install header
### install fragment @ 4 size 4

<?xs#3FFB5658?>
<?xs#3FFB5658?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### install complete
..complete
### fxRestart

<?xs#3FFB5658?>
<?xs-3FFB5658?>
x?x?x?x??xxxx?x??x?x??xx??????xxxx???x??xxxxx?x????xx?x????xx?x????????xx?x???x??x?xxx?x???x????xxx?x????xx?x?x?xxx??x????x??x?x?x?xx?x???x????x?x?x??????xxx?x?????????x?xx?x???x??????????x?x??x????x?x???x??????x?x?????x??????xx?x?x?x?x?x??x?????x???xx<?xs-00000000?>

<?xs.3FFB5658?>
<xsbug><login name="main" value="XS"/></xsbug>
```

cdctx07, 2022-03-31:
Sorry for the delay. Busy week with work.

I enabled mxTrace. I then tried running tiled-background again, but without much luck.

```
C:\Users\<name>\Projects\iot-product-dev-book\ch10-piu\tiled-background>mcrun -d -m -p esp32/moddable_two
Starting xsbug...
### fxRestart
### fxRestart

<?xs-3FFB4D40?>
### fxRestart

<?xs-3FFB4D40?>
```

So it opened xsbug, then just sat there with ### fxRestart. Nothing was happening. I then unplugged the Moddable Two and plugged it back in, which then caused the ### fxRestart then the <?xs-#ffB4D40?>. It again just sat there. So, for fun, I unplugged and plugged back in. Same cycle. Nothing is happening, just a Moddable Two saying "Ready to install apps!"

cdctx07, 2022-03-31:
Alright, last comment for tonight. I flashed the ESP32. I then reinstalled the host, which went smoothly, and the Moddable Two restarted, with debugging codes appearing in the command prompt. I then went back to tiled-background and ran mcrun...

```
C:\Users\<name>SS7\Projects\iot-product-dev-book\ch10-piu\tiled-background>mcrun -d -m -p esp32/moddable_two
Starting xsbug...
### fxRestart

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,0,0,7880,7912,3408,8192,832,4800,0,2,0,0,0</samples></xsbug>
### fxRestart

<?xs#3FFB4D40?>### set time

<?xs#3FFB4D40?>### install

<?xs#3FFB4D40?>
<?xs-3FFB4D40?>
xÇx°xÇxÇÇxxxx°xÇÇÇÇxÇ<?xs-00000000?>

<?xs.3FFB4D40?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4D40?>
<?xs.3FFB4D40?>
<set-all-breakpoints><breakpoint path="exceptions" line="0"/></set-all-breakpoints>

<?xs.3FFB4D40?>
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrument name="Frames drawn" value=" frames"/><instrument name="Network bytes read" value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><instrument name="Network sockets" value=" sockets"/><instrument name="Timers" value=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco display list used" value=" bytes"/><instrument name="Piu command List used" value=" bytes"/><instrument name="SPI flash erases" value=" sectors"/><instrument name="System bytes free" value=" bytes"/><instrument name="CPU 0" value=" percent"/><instrument name="CPU 1" value=" percent"/><instrument name="Chunk used" value=" / "/><instrument name="Chunk available" value=" bytes"/><instrument name="Slot used" value=" / "/><instrument name="Slot available" value=" bytes"/><instrument name="Stack used" value=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="Garbage collections" value=" times"/><instrument name="Keys used" value=" keys"/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser used" value=" bytes"/><instrument name="Floating Point" value=" operations"/></instruments></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,33,0,7880,7912,3392,8192,2256,4800,5,2,0,0,5</samples></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>76800,1,0,0,0,2,0,392,176,0,240752,4,0,7880,7912,3408,8192,896,4800,0,2,0,0,0</samples></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,0,0,7880,7912,3408,8192,832,4800,0,2,0,0,0</samples></xsbug>
```

That last line then just kept repeating, with nothing happening on the Moddable Two. I tried restarting it, and it just goes back to "Ready to install apps!"

phoddie, 2022-03-31:
Thanks for trying again. The debug output helps a bit. The `### install` is the start of installing the mod. It doesn't finish because it restarts instead. Your trace is different from what's expected. After a restart there is usually a `<?xs-00000000?>` which resets the debugging state. You don't have that. And you do have an instrumentation trace. That should only happen after the device has been booted for a second and should be proceeded by other things including `<instrument name="Pixels drawn" ...`.

FWIW – here's what I see (ran it on Moddable One, but should be another the same):
<details>

```console
$ mcrun -d -m -p esp/moddable_one
### restart
### fxRestart
)<?xs-00000000?>

<?xs.3FFF8AC4?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFF8AC4?>### install

<?xs#3FFF8AC4?>
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### installSpace: 16384, fragment size 3176
### install fragment @ 0 size 4

<?xs#3FFF8AC4?>
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### install fragment @ 4 size 3176

<?xs#3FFF8AC4?>
               o??
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### install fragment @ 3180 size 3176

<?xs#3FFF8AC4?>
               o??
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
.### install fragment @ 6356 size 2190

<?xs#3FFF8AC4????
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
.### update install header
### install fragment @ 4 size 4

<?xs#3FFF8AC4?>
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### install complete
..complete
### fxRestart

<?xs#3FFF8AC4?>
<?xs-3FFF8AC4?>
)<?xs-00000000?>

<?xs.3FFF8AC4?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFF8AC4?>
<?xs.3FFF8AC4?>
<set-all-breakpoints><breakpoint path="exceptions" line="0"/><breakpoint path="/Users/hoddie/Desktop/two-mods/main.js" line="67"/></set-all-breakpoints>

<?xs.3FFF8AC4?>
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrument name="Frames drawn" value=" frames"/><instrument name="Network bytes read" value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><instrument name="Network sockets" value=" sockets"/><instrument name="Timers" value=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco display list used" value=" bytes"/><instrument name="Piu command List used" value=" bytes"/><instrument name="System bytes free" value=" bytes"/><instrument name="Chunk used" value=" / "/><instrument name="Chunk available" value=" bytes"/><instrument name="Slot used" value=" / "/><instrument name="Slot available" value=" bytes"/><instrument name="Stack used" value=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="Garbage collections" value=" times"/><instrument name="Keys used" value=" keys"/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser used" value=" bytes"/><instrument name="Floating Point" value=" operations"/></instruments></xsbug>

<?xs.3FFF8AC4?>
```
</details>

My best guess is that `serial2xsbug` isn't successfully restarting the ESP32. When that happens, it just keeps running from the previous boot and confusion sets in.

Serial on Windows and the ESP32 reset circuitry can be finicky, so perhaps it is just a timing problem. [This](https://github.com/Moddable-OpenSource/moddable/blob/4998482db58a7cd216c75634e42db09835402da2/tools/serial2xsbug/serial2xsbug_win.c#L274-L282) is where serial2xsbug.c restarts the device on Windows. Try changing the `Sleep(5)` to a bigger number like `Sleep(100)`. Rebuild Moddable SDK tools as before. Then try `mcrun` again. Good luck!

cdctx07, 2022-03-31:
Thanks. I set the sleep to 100 and rebuilt. I then wiped the Moddable Two again, mcconfig'd the host (still wouldn't force restart, so I unplugged and plugged back in, which then let the serial run right). I then tried mcrun in tiled-background again...

```
C:\Users\<name>\Projects\iot-product-dev-book\ch10-piu\tiled-background>mcrun -d -m -p esp32/moddable_two
Starting xsbug...
### fxRestart

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,6,0,7880,7912,3408,8192,832,4800,0,2,0,0,0</samples></xsbug>
### fxRestart

<?xs#3FFB4D40?>### set time

<?xs#3FFB4D40?>### install

<?xs#3FFB4D40?>
<?xs-3FFB4D40?>
xÇx°xÇxÇÇxxxx°xÇÇÇÇxÇ<?xs-00000000?>

<?xs.3FFB4D40?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4D40?>
<?xs.3FFB4D40?>
<set-all-breakpoints><breakpoint path="exceptions" line="0"/></set-all-breakpoints>

<?xs.3FFB4D40?>
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrument name="Frames drawn" value=" frames"/><instrument name="Network bytes read" value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><instrument name="Network sockets" value=" sockets"/><instrument name="Timers" value=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco display list used" value=" bytes"/><instrument name="Piu command List used" value=" bytes"/><instrument name="SPI flash erases" value=" sectors"/><instrument name="System bytes free" value=" bytes"/><instrument name="CPU 0" value=" percent"/><instrument name="CPU 1" value=" percent"/><instrument name="Chunk used" value=" / "/><instrument name="Chunk available" value=" bytes"/><instrument name="Slot used" value=" / "/><instrument name="Slot available" value=" bytes"/><instrument name="Stack used" value=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="Garbage collections" value=" times"/><instrument name="Keys used" value=" keys"/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser used" value=" bytes"/><instrument name="Floating Point" value=" operations"/></instruments></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,33,0,7880,7912,3392,8192,2256,4800,5,2,0,0,5</samples></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>76800,1,0,0,0,2,0,392,176,0,240752,8,2,7880,7912,3408,8192,896,4800,0,2,0,0,0</samples></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,6,0,7880,7912,3408,8192,832,4800,0,2,0,0,0</samples></xsbug>

<?xs.3FFB4D40?>
<xsbug><samples>0,0,0,0,0,2,0,0,0,0,240752,6,0,7880,7912,3408,8192,832,4800,0,2,0,0,0</samples></xsbug>
```

Still just says "Ready to install apps!" Same thing when I try multiple- screens...

phoddie, 2022-03-31:
Thanks for the quick attempt! It definitely isn't succeeding at restarting your device. And that's what is causing the failure. I'll have to think some more about what might be going wrong.

cdctx07, 2022-03-31:
Thanks.

The programmer that came with the Moddable Two seems fidgety, and if I move my Moddable Two while plugged in, it sometimes disconnects. It seems related to how the programmer is seated. I tried using another programmer that I have for an ESP8266, but I wasn't getting far with it.

I went back to my /moddable folder and installed balls. No issues there, so it seems related to the files for the IOT book.

cdctx07, 2022-03-31:
So, on that last point, I got some header wires out to see if they help the connection between the programmer and Moddable Two more secure. Definitely can move the module around without it going black or my computer indicating a disconnection.

I then erased the ESP32 flash then mcconfig'd the host. This time it properly reset the Moddable Two and showed the serial comms. That was a plus.

I then went to mcrun the tiled-background pack. I ended up with the following this time:

```
C:\Users\coled\Projects\iot-product-dev-book\ch10-piu\tiled-background>mcrun -d -m -p esp32/moddable_two
### fxRestart
xÇx°xÇxÇÇxxxx°xÇÇÇÇxÇ<?xs-00000000?>

<?xs.3FFB4D40?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4D40?>### install

<?xs#3FFB4D40?>
<?xs#3FFB4D40?>
### fxCommandReceived: remote operation failed with resultCode 65535
NMAKE : fatal error U1077: 'C:\Users\coled\Projects\moddable\build\bin\win\release\serial2xsbug.EXE' : return code '0xffffffff'
Stop.
``` 

Any insight as to what resultCode 65535 means?

phoddie, 2022-03-31:
I'm sorry to hear you are having problems with the physical connection between the Moddable Two and programmer. I use both nearly every day without trouble. But, somehow it sounds like you got a problematic one. If you'd like us to replace the programmer, just send a note to info@moddable.com and we'll get one out to you.

Meanwhile, debugging. It does look like your device is restarting now. That's great. When install begins, the first things serial2xsbug does is request the amount of free space on the ESP8266/ESP32 for the mod. In the log I posted above, that looks like this:

```
<?xs#3FFF8AC4?>### install

<?xs#3FFF8AC4?>
<?xs#3FFF8AC4?>
### fxCommandReceived: remote operation SUCCESS with resultCode 0
### installSpace: 16384, fragment size 3176
```
In your case, instead of "remote operation SUCCESS" you have an error of 65535. That's a -1, and likely coming from the default case [here](https://github.com/Moddable-OpenSource/moddable/blob/4998482db58a7cd216c75634e42db09835402da2/xs/platforms/esp/xsPlatform.c#L1237-L1239). That would happen if the get install space command was not available, which happens when running a host that does [not support mods](https://github.com/Moddable-OpenSource/moddable/blob/4998482db58a7cd216c75634e42db09835402da2/xs/platforms/esp/xsPlatform.c#L1182-L1205). Is it possible that you aren't running the chapter 10 host, but something else? Maybe try rebuilding the host cleaning and reinstalling it, to be sure:

```
cd [chapter 10 host]
mcconfig -d -m -p esp32/moddable_two -t clean
mcconfig -d -m -p esp32/moddable_two
```

Then try `mcrun` again.

cdctx07, 2022-05-01:
Thanks, @phoddie. I have reached out.

Sorry for the delay; I got sidetracked with another short-win project and am now coming back to the Moddable Two.

phoddie, 2024-02-24:
Closing because inactive. If there's something here to address, please update this report and re-open. Thank you.

closed 2024-02-24 by phoddie