Moddable-OpenSource/iot-product-dev-book #19: Ch1 helloworld-gui - screen is undefined

KevinBatdorf, 2022-08-08:
**Build environment:** macOS

**Target device:** NodeMCU ESP32 Devkit ([amazon](https://www.amazon.com/dp/B07Q576VWZ))

**Description**

I'm going through chapter 1 and wired up the display on the ESP32. I'm attempting to run the `helloworld-gui` example.

Previously I ran `mcconfig -d -m -p esp32/nodemcu` from the ch1 hosts directory (not sure that's important). Then following the instructions I ran it again from the helloworld-gui directory.

It opened up the debugger and errored on `screen`

> helloworld-gui/main.js (20) # Break: ?: get screen: undefined variable!

**Images**

[image]

**Other information**

I'm assuming `screen` is a global that should be defined, and I'm missing a module? 

The devkit esp32 has different pins but I found a pinout sheet online and used that. I double checked but I could have made an error with that as well.

Thanks

phoddie, 2022-08-08:
The `screen` global is only present for platforms that have a display. The `esp32/nodemcu` has no screen, it is just a bare module. Some ESP32 platforms that have a display are: `esp32/moddable_two`, `esp32/m5stack`, `esp32/m5stack_fire`, `esp32/heltec_wifi_kit_32`, and `esp32/m5stick_c`. There's a good list available [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#platforms-esp32).

If you have wired a display to your ESP32 NodeMCU board following the instructions on Page 19, then you can use the `esp32/moddable_zero` target as noted near the bottom of page 22.

KevinBatdorf, 2022-08-08:
I missed that, thanks. And I confirm it's working properly using that target. Thanks for the fast reply.

closed 2022-08-08 by KevinBatdorf

phoddie, 2022-08-08:
Cool. Glad you got it working. I usually get the wiring wrong on my first three attempts. ;)

KevinBatdorf, 2022-08-08:
Yeah no joke. Me too normally ha ha. Somehow got lucky with that this time :)