Moddable-OpenSource/iot-product-dev-book #20: Invalid manifest

quangd, 2023-01-31:
The building balls worked perfect from my Moddable Two device, but it had "Invalid manifest" after running hackberry-hand app.
### Error: 'C:\Users\newuser\Projects\hackberry-hand\app\manifest.json': invalid manifest!
Question: 
1- Do I need to edit the Sys Environment from "Moddable" to "Hackberry-hand"?
2- Is there anything that I need to download to make it works?

Thanks,

phoddie, 2023-01-31:
Sorry about that. The manifest.json needed a small update to be valid JSON. If you grab the latest, it will build.

quangd, 2023-01-31:
It worked after downloading from the following link https://github.com/Moddable-OpenSource/hackberry-hand but it came up another issue.
1- Touchscreen looks like unstable.
2- the top left corner and 2nd top right corner hand control features are the same although the LCD screen shows different.

[hackberry-hand-013023.xlsx](https://github.com/Moddable-OpenSource/iot-product-dev-book/files/10543992/hackberry-hand-013023.xlsx)

phoddie, 2023-01-31:
That sound like a problem with the touch screen itself, not the hardware. You may want to verify that the touch is working. A good test for that is the Piu "drag" example:  $MODDABLE/examples/piu/drag.

quangd, 2023-02-01:
Yes, you are right. The touch screen doesn't work well after trying "drag"

closed 2023-02-01 by quangd