Moddable-OpenSource/iot-product-dev-book #22: [Bug]

ffinacio, 2023-09-19:
Build environment: Windows11

Target device: NodeMCU ESP32

**Description**
[image]

Pages 12 and 13 - Installing helloworld
The host installation was done with success.
When running mcrun -d -m -p esp32 the application stops reporting an issue involving idf-py and no access to serial2xsbug.exe and xsbug according attached report.
Best regards

phoddie, 2023-09-19:
Thanks for the report. Sorry for the trouble.

It would help to know how you installed the Moddable SDK. Did you follow the [instructions here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#win-instructions) in the Moddable SDK repository or did you use [xs-dev](https://xs-dev.js.org)? Either is fine, but it might be relevant.

I'll take a look to see if I can reproduce this.

ffinacio, 2023-09-20:
To install Modable SDK I followed the first option (moddable/documentation/Modable SDK-Getting Started.md) using Windows 11 and installed Modable seems to be working perfectly.
For my understanding, the issue can be caused by my previous installation of ESP-IDF. The explanation for "Using the Moddable SDK for ESP32" is not easy to understand and I decided to create an environment variable "ESP-IDF C:\Expressif\frameworks\esp-idf-v4.4" perhaps something else is required. As I told you before the host for Hello World was installed perfectly using mcrun.

ffinacio, 2023-09-20:
My mistake on previous comment I referred mcrun and should be mcconf.

phoddie, 2023-09-20:
@ffinacio – Thank you for the additional information. I will take a look at get back to you.

phoddie, 2023-09-22:
Apologies for the slow response. I've had to get another Windows machine set-up for development. That is taking longer than expected (still working on it!)

In reviewing your output, it looks possible that your `find` command is not the expected one. Is it possible that you have "CoreUtils for Windows" or "GnuWin32" installed? It looks like a workaround for that would be for the mcrun makefile fragment for Windows to explicitly invoke the default Windows `find` command. If you can, try changing [these lines](https://github.com/Moddable-OpenSource/moddable/blob/c5845e51c450fd45cb382a65422bf6298e29970d/tools/mcrun/nmake.esp32.mk#L52-L55
) to:

```
debug: $(ARCHIVE)
	-tasklist /nh /fi "imagename eq serial2xsbug.exe" | ($(SYSTEMROOT)\System32\find.exe /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
	tasklist /nh /fi "imagename eq xsbug.exe" | $(SYSTEMROOT)\System32\find.exe /i "xsbug.exe" > nul || (start $(XSBUG).exe && echo Starting xsbug... && timeout /nobreak /t 7 > nul)
	echo "Running serial2xsbug..." && set "XSBUG_PORT=$(XSBUG_PORT)" && set "XSBUG_HOST=$(XSBUG_HOST)" && serial2xsbug $(UPLOAD_PORT) $(DEBUGGER_SPEED) 8N1 -install $(ARCHIVE)
```