Moddable-OpenSource/iot-product-dev-book #24: [Bug] Cannot run `ch3-network/http-get` example

achou11, 2024-02-24:
I'm trying to run example in https://github.com/Moddable-OpenSource/iot-product-dev-book/tree/master/ch3-network/http-get but I'm running into a couple of issues:

1. Running a build using `mccrun` gives me the following error:

```sh
Installing mod.### fxCommandReceived: remote operation id 0xff0f failed with resultCode -8
make: *** [debug] Error 255
```

2. When I try to build via `mcconfig`, I get the following error:

```sh
WARNING: No 'creation' found in manifests.
   Did you intend to build using mcrun?
### Error: project manifest or included manifest must include creation
```

Figured that maybe I need to adjust the manifest so that it includes `"$(MODDABLE)/examples/manifest_base.json"` since the previous examples seemed to have it. So added that and ran `mcconfig` again, but then I get the following error in xsbug:

```sh
# Break: (host): module "http" not found!
```

---

Installed the Moddable SDK via xs-dev. Here's the output for `xs-dev`:

```sh
xs-dev environment info:
  CLI Version                0.30.3                                                                                   
  OS                         Darwin                                                                                   
  Arch                       x64                                                                                      
  Shell                      /bin/bash                                                                                
  NodeJS Version             v20.11.0 (/Users/andrewchou/Library/Caches/fnm_multishells/87784_1708746403799/bin/node) 
  Python Version             3.12.2 (/Users/andrewchou/.local/share/mise/installs/python/3/bin/python)                
  Moddable SDK Version       4.4.0 (/Users/andrewchou/.local/share/moddable)                                          
  Supported target devices   mac, esp32                                                                               
  ESP32 IDF Directory        /Users/andrewchou/.local/share/esp32/esp-idf  
```

achou11, 2024-02-24:
Ah wait I just realized that I didn't read page 137 carefully enough about building the host specified in `ch3-network/host` and then running the examples 🤦‍♂️ sorry for the noise!

closed 2024-02-24 by achou11

phoddie, 2024-02-24:
Here's the correct sequence of steps:

```
cd iot-product-dev-book (wherever you've checked out the book's git repo)
cd host
mcconfig -d -m -p esp32
// once it starts running do Control C, then...
cd ../http-get
mcrun -d -m -p esp32 ssid="YOUR WIFI" password="YOUR PASSWORD"
```

Chapter 1, Getting Started, explains that the process is to first install the host for the chapter using `mcconfig`, and then run the example with `mcrun`.

Good luck~

phoddie, 2024-02-24:
Congrats on figuring that out!